nightscout/AndroidAPS #2524: [DEV] DASH: Temp basal cancelled/stopped by pod for unkown reason, AAPS History not updated

jbr7rr, 2023-04-30:
Build: 34e7a9ad52-dev-i (personal branch) merged at 6cfd5b0c1673a79d9d33498ee2a7031b54ccd6c4
Pump: Dash
Time: Temp basal start at 11:52:37.814. Reconnect where we find the changed deliveryStatus from deliveryStatus=TEMP_BASAL_ACTIVE to deliveryStatus=BASAL_ACTIVE at 17:04:08.902
Logs: [AndroidAPS_LOG_1682783403444.log.zip](https://github.com/nightscout/AndroidAPS/files/11361767/AndroidAPS_LOG_1682783403444.log.zip)

Context: On this day I went to the pool, set a 10 hour long basal at 11:52, did a few bolus after that, but did not change the basal or profile. The phone was stored in the locker so had no contact with the pod for some time, at 17:00 the pool closed so I went home. In the car I noticed my dash ticking while AAPS reported still a 0 temp basal. so I checked the status screen. found this:

[image] [image]

As you can see the basal was delivering, but it somehow still stated there was a TBR active. even though the deliveryStatus of the pod was BASAL_ACTIVE. I have not received any alarm other then no BG and no pump connection.

I see 2 problems here:

1. Why did the pod stop the TBR? I have no real answer to this other then some guesses: 
    a. maybe the pod got damaged during the wild water slide, the pod was pretty banged up and had some faint yellow stains inside. 
    b. Or maybe the pod cancels the TBR when the controlling device is out of range for x ammount of time? I intent on testing this hypotheses

2. The pod rapported it had not TBR active anymore. I expect the history and tbr status to be updated, this did not happen. However I did dive a bit into this part of the dash code and the status message, it is hard to know at what time the TBR was stopped, but we can make an educated guess by comparing the delivered insulin. I can implement something for that

Any other guesses/input? Especially on why the pod had stopped the TBR?

vanelsberg, 2023-04-30:
@jbr7rr 

> **The phone was stored in the locker so had no contact with the pod for some time**

I think this is by design.
The Pod will always fall back to it programmed basalrate when it does not received communication for 30 minutes.

So what you describe is expected behaviour when using the PDM or AAPS.

jbr7rr, 2023-05-01:
> @jbr7rr
> 
> > **The phone was stored in the locker so had no contact with the pod for some time**
> 
> This is by Insulet design for both EROS and DASH. For safety reasons the Pod will always fall back to it programmed basalrate when it does not received communication for 30 minutes.
> 
> So what you describe is expected behaviour when using the PDM and also AAPS.

I did some digging in the manual:
https://www.omnipod.com/sites/default/files/2021-04/Omnipod-DASH_User-Guide_English.pdf
----------------
On page 15:
Note: Even when there is no
communication, the Pod continues
delivering insulin according to the
instructions it was given. See "What the
Pod can do between PDM instructions"
on page 148 for details.

on page 148:
What the Pod can do between PDM instructions
Once activated, the Pod can perform the following **without input** from the PDM:
• Deliver insulin according to the active Basal Program's schedule. The Pod
has a built-in clock that allows it to track the timing of the Basal Program's
segments.
**• Stop delivery of a temp basal at the scheduled time and resume delivery of the
active Basal Program**.
• Once a bolus or an extended bolus is started, complete the delivery of the
bolus.
• Track the amount of insulin remaining inside the Pod.
• Track the amount of time until Pod expiration. Stop delivering insulin once
the Pod expires.
• Perform self-checks to verify that insulin delivery is occurring as expected.
Sound an alarm if it is not.
• Send alarm and notification messages to the PDM, if it is in range.
• Stop insulin delivery if the Auto-off feature is enabled and you have not used
the PDM within the specified period of time.
Auto-off
The optional Auto-off feature is useful if you are prone to hypoglycemia
unawareness. When you enable the Auto-off feature, you define a length of time
for an Auto-off countdown timer. The Pod and PDM will sound an alarm if you
do not use your PDM within that amount of time. To change the Auto-off setting,
see "Pod Auto-off" on page 109.

-------------

There is no stopping of the temp basal mentioned anywhere else. and according to this insulin delivery should continue as programmed. The Auto-off feature is hardcoded to be disabled in the DASH driver if I'm correct (I will check this). If you have any other source saying something else, please let me know

Given this I would say that the behavoir in AAPS is not as expected Note: On 2 points, as mentioned in my top post. 1: TBR stopped early, 2. AAPS not recognizing that the TBR is stopped (I updated the main post to clarify this). 

Also the times of events do not allign:
```
- 11:52 Start 0 Temp for 600 minutes
- 14:14 Last connection with POD
- ~15:55 TBR Stopped by POD
- 17:05 Regained connection with POD, Home screen stil showing 0 Basal, while DASH tab states BASAL_ACTIVE
```

What I find interesting here is that between start TBR and stop TBR by POD is roughly 4 hours. The time between last connection and stop TBR Is 1:41 which is not a rounded number and does not make sense to me for the POD to do that purely on no connection based on the above from the manual

vanelsberg, 2023-05-01:
>  17:05 Regained connection with POD, Home screen stil showing 0 Basal, while DASH tab states BASAL_ACTIVE

Did you push the refresh button? AAPS needs to get the status from the Pod. It is not refreshing by itself and afaik will only update after executing a command.

> - 11:52 Start 0 Temp for 600 minutes

How does one set e 0-temp for 600 minutes?
Or do you mean "**suspend loop**"?
I think suspend loop just cancels any running TBR and stops sending commands (so Pod falls back to programmed basal rates)

> - ~15:55 TBR Stopped by POD
> "calculated by tracing back the delivered insulin based on my basal schedule"

Initial TBR was started at 11:52, repeating every 30 mins. So last TBR "repeat" command was at approx 13:52h. From there no connection - so this last TBR command ran for 120 minutes and stopped at 15:52h, falling back at programmed basal rate.

> There is no stopping of the temp basal on loss of connection mentioned anywhere.

You may be right on that as I can not find any mention on this in the DASH manual. So maybe then this is no longer true for DASH or I was completely wrong on this and this is driver functionality.

jbr7rr, 2023-05-02:
> AAPS Overview/Homescreen is know to lag behind, only updating when a new BG is received triggering oRef calculations. Also note one needs to push the refresh button. AAPS is not auto refreshing the Pod status and will only update after executing a command.

See screenshots above, these are way after first reconnect with DASH, and yes I did hit refresh button multiple times after I suspected something was wrong. But this did not change. Also history in NS did not show the TBR cancalled

> How does one set a 0-temp for 600 minutes? I do not think that is possible - not even through a basal setting in the profile. Or do you mean "**suspend loop**"? I think suspend loop just cancels any running TBR and stops sending commands (so Pod falls back to programmed basal rates) So this is not a 0-Temp?

See "Steps to reproduce" below
Yes it is a 0 Temp Basal, as in 0U/h for 600 minutes

> Initial TBR was started at 11:52, repeating every 30 mins. So last TBR "repeat" command was at approx. 13:52h. From there no connection - so this last TBR command ran for 120 minutes and stopped at 15:52h, falling back at programmed basal rate.

This is wrong, It was/is a single TBR command for setting a TBR of 600 minutes, AAPS does not segment it as far as I know and can see from logs and code.

> You may be right on that and there is no mention on this in the DASH manual. So actually it is not the Pod... Confused: But the 30mins/120mins mentioned above ensures a programmed TBR will _always end_ after max 120mins in case AAPS can not connect to the Pod to stop or reschedule...

This only applies when loop mode is active, In manual mode it is a different story and the user can do whatever he wants. Also keep in mind that the pumpcontrol apk is freely downloadable and quite some dash users seem to use that as a starting point. Meaning if this is not a one time incident it could be a larger problem

jbr7rr, 2023-05-02:
I was able to reproduce it on my test phone using the following steps:

# Steps to reproduce:

- Deactivate loop (theoretically this also effects pump control apk, not tested yet)
- In action tab start temp basal: 0U/h for 600 min
- Check DASH tab: Delivery Status should read: TEMP_BASAL_ACTIVE and Temp basal active 0.00U/h
- Keep phone close to pod, but no further input into AAPS
- After 4:15 :
    - Refresh from DASH tab
    - Expected behavior: Delivery Status: TEMP_BASAL_ACTIVE and temp basal active 0.00U/h and pod not delivering insulin
    - Actual behavior: Delivery Status: BASAL_ACTIVE and temp basal active 0.00U/h and pod is delivering insulin

Screenshots of test:
[image] [image] [image]

vanelsberg, 2023-05-02:
Interesting. Curious on what the Pod history shows. Any command send to the Pod should show up there.

vanelsberg, 2023-05-02:
Interesting details: original DASH beta/dev driver had a separate suspend button on the DASH tab. When activated it cacled all running TBR's and blocked any Pod command until unsuspended.

Shortly after merge/release to main AAPS dev it was removed arguing this to be functionally identical to the AAPS suspend mode and only confusing users.

vanelsberg, 2023-05-02:
I do not think "manual" usage as described above was covered nor tested for during DASH development.

@MilosKozak This could lead to unexpected and possible dangerous situations where the user is not aware of the Pod delivering insulin while setting a long duration 0-TBR.

As I have no insight on a possible way to fix, I would propose the following:
1. Warn dev users
2. Until this is fixed, disable the option to manually set a TBR (activity button) when in suspend mode (maybe only enable for developer mode)

@jbr7rr **Good find!** 
Interested in your view on this!

jbr7rr, 2023-05-03:
> I do not think "manual" usage as described above was covered nor tested for during DASH development.
> 
> @MilosKozak This could lead to unexpected and possible dangerous situations where the user is not aware of the Pod delivering insulin while setting a long duration 0-TBR.
> 
> As I have no insight on cause or a possible way to fix, I would propose the following:
> 
>     1. Warn dev users
> 
>     2. Until this is fixed, disable the option to manually set a TBR (activity button) when _loop is disabled_ (maybe only enable for developer mode)
> 
> 
> @jbr7rr **Good find!** Interested in your view on this!

I think that this issue also applies for master, I did not test, but not much has changed in these parts of the code since then. So we need to take that into account as well.

A proper fix needs more investigation, I will do this in the coming weekend.
If a proper fix cannot be implemented soon, or requires lots of testing. I will propose a hotfix by limiting the TBR range to 4 hours (Will need to test if that works properly as well), This is easier to implement then disabling manual TBR.

avereha, 2023-05-25:
The duration for this command is specified in minutes. 
4:30 is very close to 256(minutes). Maybe we don't serialize the command properly if time > 1 byte.

avereha, 2023-05-25:
Maybe we don't serialize the tempBasal correctly, will try to check this in the next days.

jbr7rr, 2023-05-25:
> Maybe we don't serialize the tempBasal correctly, will try to check this in the next days.

Hey! I think the serialization is not the problem, as Loop (iOS) had the same problem, I implemented a work around for the core issue in https://github.com/nightscout/AndroidAPS/pull/2531 which has been tested exenstivly, and with that fix all basal rates works for all TBR rates

closed 2023-06-01 by jbr7rr