nightscout/AndroidAPS #2525: Dev dynISF Enhancement: visualization of ISF

yhonatanb20, 2023-04-30:
3.2.0 Dev-i 27.4.23
Is it possible to create a visualization of the canges in ISF made by the dynISF plugin on the overview screen graphs, i.e. like TBRs or sensitivity are visualized?
Tank you all dynISF devs :) 💯

DM-LTR, 2023-05-04:
It is possible via NS API for example... Data taken from NS/api/properties/, read by python script, sent to mqtt, saved to influx and viewed by Grafana
[image]
[image]

PieterGit, 2023-09-11:
@DM-LTR : can you share your script?

DM-LTR, 2023-09-14:
Script author will provide his solution here soon.

xmizi, 2023-09-14:
This ISF visualisation is a part of my Grafana dashboard. All sources (old python script, node-red flow + grafana dashboad) are possible to download from  https://github.com/xmizi/dm1-tools.

justmara, 2023-09-21:
ergh, this is here for a year already 
https://github.com/justmara/ns-exporter

MilosKozak, 2026-10-11:
AAPS can show the variable ISF (sensitivity) as a series in the overview graphs. Open the graph settings and add "Variable sensitivity" to a secondary graph.

closed 2026-10-11 by MilosKozak