nightscout/AndroidAPS #2529: Having separate SMB "profiles"

Etti2020, 2023-05-04:
Is it possible to have different SMB "profiles", or even just enable/disable SMB at different times of the day?
E.g. for a little child, we'd prefer to disable the SMB option at night only, or change its parameters so that it is less aggressive than we need it to be during the day...
Also, it would be amazing to have the same with sound alerts - e.g. turn them off during the day or specific occasions (when she's embarrassed to be "beeping" in publc) but on at night (so we'd wake up if her BG starts dropping). Right now this has to be done manually - turned on and off - and runs the risk of forgetting to turn it on at night. 
A safety feature can leave just the super-urgent notifications on.
Thanks!!!!

olorinmaia, 2023-06-14:
Second on this improvement. Would love the option to configure SMB-basal-minutes and UAM for spesific time periods. Example is when kid is starting sleeping bigger SMB's can be allowed then rest of the day, and during night smaller smb's can be configured.

szantos, 2023-07-20:
I'd also wish an automation action for disabling SMBs, but kindly welcome any other finer-graded enhancement. God-mode would be the ability to set any of the preferences with an automation action based on an automation trigger...

A workaround (with flaws) for disabling SMBs for some periods of the day:
- in preferences: don't allow SMB with high TT
- set an automation with high TT at times you want

The flaws:
- every high TT will be affected
- the minimum high TT could be too high for some (can be changed in code simply)
- you have to carefully check automation prerequisites and automationorder not to ruin other settings

Kenoubi, 2023-08-06:
> the minimum high TT could be too high for some (can be changed in code simply)

I actually think it's the opposite.  101 mg/dl is not a "high" temp target at all; that's not even high enough to avoid hypos reliably if one suddenly needs to move around fast or lift heavy things, etc.  But yes, if you're willing to make any changes to the code at all, that is an easy change to make compared to most others.