nightscout/AndroidAPS #2530: Bolus given, but not taken into account

Keksbendiger, 2023-05-04:
- SMBs and possibly whole meal boluses are being injected, but do not appear in the treatments history, nor are they added to IOB, which results in an **endless loop of pumping in more and more insulin**
- in the combo tab the correct data is shown as "last bolus ...", TBR are shown correclty everywhere
- seems to be like that since starting objective 9
- Accu Chek Spirit Combo
- occured in 3.0.0.2 and still persists after updating to 3.1.0.3
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/11401832/AndroidAPS.log)
[AndroidAPS._2023-05-04_22-56-39_.0.log](https://github.com/nightscout/AndroidAPS/files/11401843/AndroidAPS._2023-05-04_22-56-39_.0.log)
[AndroidAPS._2023-05-04_00-00-02_.16.log](https://github.com/nightscout/AndroidAPS/files/11401844/AndroidAPS._2023-05-04_00-00-02_.16.log)

jotomo, 2023-05-08:
According to the logs the pump was changed at 04.05.23, ~23:02.
The combo pump is identified by the MAC address which changed from eKxys3bvBxs (???) to 00:0E:2F:69:C3:6D.
To avoid importing boluses all boluses before the date of a pump switch are ignored.

From what I gather, the first SMB (0.9) was incorrectly NOT added to treatments. Successive boluses and SMBs should work well.

@Keksbendiger Did you change the pump, upgrade the OS, toggle MAC randomization or something of the sort?

Keksbendiger, 2023-05-08:
No I did not change the pump.
Also having that problem #2017 all the time.
Instead of regular ~150 Units it pumped over 250 in one day.

dv1, 2023-11-16:
Late to this, but: Does this still happen in 3.2.0.2 for you? Did you try the new combov2 driver? Does it happen with that one too?

Keksbendiger, 2023-11-17:
> Late to this, but: Does this still happen in 3.2.0.2 for you? Did you try the new combov2 driver? Does it happen with that one too?

Thanks for checking out! Have not updated yet, but will do in the next few days. Gonna re-enable SMB and try it out :)

Keksbendiger, 2023-12-12:
Updated to 3.2.0.2 and re-enabled all SMB options.
Also switched to comboV2 driver.

Encountered no problems regarding SMB or bolus history, all SMBs seem to be listed under "Treatments".
TDD did also not increase.

Would say the ticket can be closed as fixed in 3.2.0.2

closed 2023-12-13 by MilosKozak