nightscout/AndroidAPS #2538: Set 120% for 20 min but see negative minutes

robertrub, 2023-05-12:
Last build
[image]

There is a bug somewhere but can't get my hand on how to trigger it. I had set profile at 120% for 20 min. Some time later, I see -minutes and start another 120% for 20 min using the button _use 120%_ some time later, I see that I have more -minutes and set 120% for 20 minutes again but I have no indication that the command was accepted.

Last dash connection less than 1 min ago.

[image]

When I look in the treatments, the commands have been accepted.

[image]

Here is the logs. You have the times on the capture above.

[image]

robertrub, 2023-05-13:
Might be related to reactivating a temp profile when the actual one is not over yet.

MilosKozak, 2023-05-13:
fixed by https://github.com/nightscout/AndroidAPS/commit/01bb554651f2dbfafdec7f069dd99f2ed16cc578

closed 2023-05-13 by MilosKozak