nightscout/AndroidAPS #2543: Androidaps does not display latest value as read by xdrip

01erster, 2023-05-16:
On AndroidAPS Overview Tab the reading time of blood sugar value is behind xdrip+ latest reading. I have recorded this for 14.05.2023 11:06

Log files and screen recording in attached files:
[AndroidAPS_LOG_1684055313489.log.zip](https://github.com/nightscout/AndroidAPS/files/11489655/AndroidAPS_LOG_1684055313489.log.zip)
[VID_20230514_111241.zip](https://github.com/nightscout/AndroidAPS/files/11489658/VID_20230514_111241.zip)

MilosKozak, 2023-05-17:
expecting this is not 3.2.0?

01erster, 2023-05-17:
It is this version:
[image]

MilosKozak, 2023-05-18:
this code is completely new in 3.2.0, test beta or dev or wait for next release

closed 2023-05-18 by MilosKozak