nightscout/AndroidAPS #2552: EOPatch/GlucoMen Day PUMP:

jbm76, 2023-05-23:
Hello,

This morning I have to change the patch after reaching 3 and a half days but still not reaching the fourth day.
I started the process, step 1, and AAPS told me that it was going to stop and cancel my basal, I accepted this information.

I decided not to immediately apply the new patch, and that was when I saw that on the AAPS home screen, it showed basal rate although it seemed disconnected (both insulin and treatment icons were missing).
Build 
Report
/
-------
<details><summary>Details</summary>
<p>

</p>
</details> ```

``` -------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

fjpezuela, 2023-06-09:
Seems like it happens to all of us.

When the patch is deactivated and none is active, the AAPS main screen shows the basal for that time (even though we don't have any on). I don't know if it's an aesthetic failure or if he takes it into account in his calculations.

MilosKozak, 2026-10-14:
Fixed in `dev` by 3edc1ec3bd.

Without an active patch, AAPS now treats the pump as suspended and records a 0 U/h temp basal for the gap. The overview no longer shows a basal rate between patches, and the IOB no longer counts basal insulin that was never given. When the new patch is activated, the temp basal ends and the normal basal runs again.

Tested with a real patch: deactivation and then activation of a new patch.

closed 2026-10-14 by MilosKozak