nightscout/AndroidAPS #2553: Random Alarm Sound

PythonFZ, 2023-05-24:
I started using AAPS with Virtual Pump and MDI. I'm getting alarm sounds I can not identify or deactivate. It is just an alarm without any further notification. I searched the settings and disabled everything that looks like an alarm to me. 

Are there alarms that can not be turned off?
I'm using 14.2.6 on a Pixel 7. Data comes from xDrip and NS is connected.

yuri-vashchenko, 2023-05-24:
I am also getting some unwanted sound that I would like to be able to turn off or disable.
I'm not sure we are talking about the same alarm sound, but mine is clearly related to Nightscout.
It always plays when my phone switches from mobile data connection to Wi-Fi or vise versa (or when my phone gets new IP address). If I pause the nightscout (select corresponding checkmark) in NS client tab in AAPS, the sound stops playing every time my phone is (re)connected to the internet.

PythonFZ, 2023-05-24:
I can reproduce @yuri-vashchenko alarms. It is the same alarm sound for me. I don't know for sure if it only happens on new connections.

MilosKozak, 2023-05-24:
disconnect NS to see if origin is AAPS

PythonFZ, 2023-05-24:
The Alarm occurs reproducibly when:
- NS Client restart
- connecting wifi
- after un-pausing NS client

The NS Client is configured to only use WiFi connections.

MilosKozak, 2023-05-24:
then I'd say it's comming from NS. you can disable NS alarms in preferences

PythonFZ, 2023-05-24:
Both alarms are turned off for nightscout without an effect.

jbr7rr, 2023-05-24:
If you are using xDrip+ please check this:
https://navid200.github.io/xDrip/docs/AAPS_ProfileImportNotification.html

PythonFZ, 2023-05-24:
@jbr7rr thanks you very much! And thank you @MilosKozak for the great work as well!

closed 2023-05-24 by PythonFZ