nightscout/AndroidAPS #2564: Profile timezone in Profile Switch treatments sent to Nightscout is off-by-one

Jackenmen, 2023-06-01:
It seems that Profile Switch treatments created by AndroidAPS have incorrect timezone in the `timezone` field of the JSON included in the `profileJson` field. More specifically, the timezone seems to be off-by-one - for me, that means that I get `Africa/Cairo` (UTC+3) rather than my actual timezone which is `Europe/Warsaw` (UTC+2).

The relevant Nightscout API call where I've noticed that Nightscout returns the wrong data is:
```
https://[nightscout-url]/api/v1/treatments.json?find[eventType][$eq]=Profile%20Switch&count=1
```

Relevant log: [AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/11621304/AndroidAPS.log)

I'm getting this on build version: 1de236a-2023.05.30-23:35
Both my Nightscout profile timezone and my phone's timezone are set to Europe/Warsaw.
I'm currently on a virtual pump with NSClient providing the BG data.

MilosKozak, 2023-06-01:
are they shifted on NS graph too?

Jackenmen, 2023-06-01:
It's only about the Profile Switch treatment having incorrect "profileJson" - "created_at" field of the treatment (and therefore the position of the treatment on the graph) is correct. I'm not sure if there's something else that could appear shifted but I can't really see anything that is. My problem is with the incorrect data being stored in the treatment which is problematic when working with the data from the API.

MilosKozak, 2023-06-02:
can you provide example of such wrong json?

Jackenmen, 2023-06-02:
Huh, I thought I already did, sorry about that... Here's how the output from the `https://[nightscout-url]/api/v1/treatments.json?find[eventType][$eq]=Profile%20Switch&count=1` endpoint that I mention in issue description looks:

<details>
<summary>Full output from the API endpoint</summary>

```json
[
    {
        "_id": "64786280c846250208181aca",
        "timeshift": 0,
        "percentage": 100,
        "profile": "Workday",
        "originalProfileName": "Workday",
        "created_at": "2023-06-01T09:18:23.000Z",
        "enteredBy": "openaps://AndroidAPS",
        "isValid": true,
        "eventType": "Profile Switch",
        "duration": 0,
        "profileJson": "{\"units\":\"mg\\/dl\",\"dia\":5,\"timezone\":\"Africa\\/Cairo\",\"sens\":[{\"time\":\"00:00\",\"timeAsSeconds\":0,\"value\":80},{\"time\":\"05:00\",\"timeAsSeconds\":18000,\"value\":50},{\"time\":\"22:00\",\"timeAsSeconds\":79200,\"value\":80}],\"carbratio\":[{\"time\":\"00:00\",\"timeAsSeconds\":0,\"value\":6.2},{\"time\":\"05:00\",\"timeAsSeconds\":18000,\"value\":4.7},{\"time\":\"10:00\",\"timeAsSeconds\":36000,\"value\":4.7},{\"time\":\"13:00\",\"timeAsSeconds\":46800,\"value\":5.5},{\"time\":\"15:00\",\"timeAsSeconds\":54000,\"value\":6}],\"basal\":[{\"time\":\"00:00\",\"timeAsSeconds\":0,\"value\":0.15},{\"time\":\"01:00\",\"timeAsSeconds\":3600,\"value\":0.2},{\"time\":\"02:00\",\"timeAsSeconds\":7200,\"value\":0.3},{\"time\":\"03:00\",\"timeAsSeconds\":10800,\"value\":0.5},{\"time\":\"04:00\",\"timeAsSeconds\":14400,\"value\":0.6},{\"time\":\"07:00\",\"timeAsSeconds\":25200,\"value\":0.7},{\"time\":\"08:00\",\"timeAsSeconds\":28800,\"value\":0.85},{\"time\":\"09:00\",\"timeAsSeconds\":32400,\"value\":0.8},{\"time\":\"11:00\",\"timeAsSeconds\":39600,\"value\":0.7},{\"time\":\"14:00\",\"timeAsSeconds\":50400,\"value\":0.35},{\"time\":\"20:00\",\"timeAsSeconds\":72000,\"value\":0.3},{\"time\":\"22:00\",\"timeAsSeconds\":79200,\"value\":0.2},{\"time\":\"23:00\",\"timeAsSeconds\":82800,\"value\":0.1}],\"target_low\":[{\"time\":\"00:00\",\"timeAsSeconds\":0,\"value\":100},{\"time\":\"07:00\",\"timeAsSeconds\":25200,\"value\":100},{\"time\":\"22:00\",\"timeAsSeconds\":79200,\"value\":100}],\"target_high\":[{\"time\":\"00:00\",\"timeAsSeconds\":0,\"value\":130},{\"time\":\"07:00\",\"timeAsSeconds\":25200,\"value\":130},{\"time\":\"22:00\",\"timeAsSeconds\":79200,\"value\":130}]}",
        "carbs": null,
        "insulin": null
    }
]
```

</details>

<details>
<summary>Contents of the `profileJSON` field</summary>

```json
{
    "units": "mg/dl",
    "dia": 5,
    "timezone": "Africa/Cairo",
    "sens":
    [
        {
            "time": "00:00",
            "timeAsSeconds": 0,
            "value": 80
        },
        {
            "time": "05:00",
            "timeAsSeconds": 18000,
            "value": 50
        },
        {
            "time": "22:00",
            "timeAsSeconds": 79200,
            "value": 80
        }
    ],
    "carbratio":
    [
        {
            "time": "00:00",
            "timeAsSeconds": 0,
            "value": 6.2
        },
        {
            "time": "05:00",
            "timeAsSeconds": 18000,
            "value": 4.7
        },
        {
            "time": "10:00",
            "timeAsSeconds": 36000,
            "value": 4.7
        },
[... 277 lines cut ...]
                        "time": "07:00",
                        "timeAsSeconds": "25200",
                        "value": "100"
                    },
                    {
                        "time": "22:00",
                        "timeAsSeconds": "79200",
                        "value": "100"
                    }
                ],
                "timezone": "Europe/Warsaw",
                "units": "mg/dl"
            }
        },
        "created_at": "2023-06-02T00:37:14.642Z",
        "defaultProfile": "Workday",
        "key600": "[REMOVED FROM THE OUTPUT]",
        "mills": "1654130228254",
        "startDate": "2022-06-02T00:37:08Z",
        "units": "mg/dl"
    }
]
```

</details>

MilosKozak, 2023-08-03:
I checked it and it doesn't have good solution because data in AAPS is stored as an offset to UTC. it's mapped to TZ string only because of NS requiremt.
I'd recommend use only unix timestamp and ignore zones at all

closed 2023-08-03 by MilosKozak

Jackenmen, 2023-09-10:
> I checked it and it doesn't have good solution because data in AAPS is stored as an offset to UTC. it's mapped to TZ string only because of NS requiremt.

The problem here was that it's off by one hour (right now, `Africa/Cairo` is UTC+3, `Europe/Warsaw` is UTC+2), not so much that it's mapping to the wrong timezone name. I decided to look into it and apparently, this is an external problem - all the calculations are correct *but* `Africa/Cairo`'s offset on my phone was 2 hours instead of 3 hours which, toppled with it being the first matching item in the collection, ended up with AAPS choosing this timezone even though it's incorrect:
[image]

The fix was to perform Google Play security update which, for whatever reason, my phone doesn't notify me about...

I do feel it would be better to instead map to `Etc/GMT*` timezones, e.g. `Etc/GMT-2` for UTC+2, `Etc/GMT-3` for UTC-3, and `Etc/GMT` (or `Etc/UTC`) for UTC since it would be less confusing but in the end, it doesn't really matter for me since with the updated tzdata, the timezone on the profile switch now at least gets set to a UTC+2 timezone, even if it's `Africa/Mbabane` rather than `Europe/Warsaw`.