nightscout/AndroidAPS #2566: MINOR: Hearth rate chart starts at 0

robertrub, 2023-06-03:
Not really a bug. The HR chart has the minimum set to 0... Not very useful :)

I think setting the min of the HR chart to 40 should cover 99,9% of cases and give a better definition of the variation of the HR.

[image]

buessow, 2023-06-27:
My heart rate, or at least what the watch measures goes below 40. So how about 30?

robertrub, 2023-06-27:
40 is real low (usually very long distance runners or long distance swimmers...). Ate you one of them ? Did you count your heart beat with a finger on the wrist and a timer and compare it to the SmartWatch ?

buessow, 2023-07-08:
I think my resting heart rate is 45 and the lower values are artefacts off
the watch.

MilosKozak, 2023-08-24:
https://github.com/nightscout/AndroidAPS/commit/ea8a58ed27834896361b1859a3f10758d510dfec

closed 2023-08-24 by MilosKozak