nightscout/AndroidAPS #2573: [DEV 3.2.0-dev-j cd9f5e4] Problem opening preferences

fjpezuela, 2023-06-05:
When opening any option that accesses anything related to preferences, the application crashes and closes. I haven't had any problems until today. I attach the logs and video.

The last error is around 20:59 (see video)

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/11655834/AndroidAPS.log)

https://github.com/nightscout/AndroidAPS/assets/67511778/5c188225-4f53-4f0b-8a06-1b630c0c64bc

Thanks.

jbm76, 2023-06-08:
Good morning,

The exact same thing happened to me yesterday. AAPS crashed and I was unable to access the preferences, just as shown in the video.

AAPS  dev-j compiled 01/06/2023

Attach Log de AAPS ( 20:45) and screenshot of how it continuously started AAPS and when trying again to access the preferences it crashed

Thanks in advance

[AndroidAPS._2023-06-07_21-08-23_.21.zip](https://github.com/nightscout/AndroidAPS/files/11688308/AndroidAPS._2023-06-07_21-08-23_.21.zip)

[image]

parapenT1sta, 2023-06-08:
Do you have the same issue if the language is set to English? I can see that both of you aren't using English as default language.

fjpezuela, 2023-06-08:
> Do you have the same issue if the language is set to English? I can see that both of you aren't using English as default language.

It could be. The problem is that we can't access the preferences to change it ;)

parapenT1sta, 2023-06-09:
> > Do you have the same issue if the language is set to English? I can see that both of you aren't using English as default language.
> 
> It could be. The problem is that we can't access the preferences to change it ;)

If you had selected "System Default", change your phone language and check if AAPS default language is changed.

[image]

fjpezuela, 2023-06-09:
I've tried, but I must have it set to "Spanish". Thanks.
> If you had selected "System Default", change your phone language and check if AAPS default language is changed.
>
> [image]

parapenT1sta, 2023-06-10:
@MilosKozak Can you check if Spanish translation is OK?

fjpezuela, 2023-06-10:
Yesterday I updated to the latest build with the latest modifications and now it opens correctly.

parapenT1sta, 2023-06-10:
Great. Now you can close the ticket.

jbm76, 2023-06-10:
Hello,

This keeps failing even with the latest version dev of APPS . ( f1118e2)

After installing it two days ago, on the third day and having "system default" selected as the language in the application, it has broken again in exactly the same way, not allowing access to the preferences menu.

Attached video identical to the previously uploaded as well as the log of the app

Please,reopen the ticket because this is failing in latest versions of the dev.

thank you very much in advance

AAPS dev - j (f1118e2) 

[AndroidAPS._2023-06-10_21-48-53_.23.zip](https://github.com/nightscout/AndroidAPS/files/11713372/AndroidAPS._2023-06-10_21-48-53_.23.zip)

https://github.com/nightscout/AndroidAPS/assets/45192678/6c8a18f1-ef34-4fae-9acf-0956a78875ef

scastanocaro, 2023-06-11:
Good afternoon

I have the same problem with the version I have just compiled.  I can access neither to preferences nor to configuration. When I select one of both functionalities AAPS crashed.

Thank you very much in advance

fjpezuela, 2023-06-12:
Hello,

I will try to add more information, after talking to more people who have the same thing happening to them:

AAPS works correctly, but days later this problem starts to occur.

In some cases, updating the version of AAPS, fixes the problem temporarily and days later the problem is reproduced again.

Reverting to a previous version and upgrading back to the latest version sometimes solves the problem temporarily.

Everyone I talk to has EOPatch2. I asked if it happens to anyone else using another pump on Discord and no one seems to have the problem.

All the people that I know with the problem, we are in Spain (I don't know if it has something to do with the language, although I left configured "System Default" and it kept happening.

If you need us to provide some more log or to do some concrete test, please, let us know.

parapenT1sta, 2023-06-12:
Do you have the same issue if using AAPS with English language only?

fjpezuela, 2023-06-12:
> Do you have the same issue if using AAPS with English language only?

I upgraded again and left the option set to use the system language setting. After the update it is working fine for me. @jbm76 switched to English and it didn't fix the problem for him, so I think it's not language related.

@jbm76 can surely explain you better.

jbm76, 2023-06-12:
@parapenT1sta 
@fjpezuela 

I had the error with:

Telephone in English -- In AAPS selected system default, therefore with AAPS in English
Telephone in Spanish--In AAPS selected system default, therefore with AAPS in Spanish.

Now I am testing with Telephone in Spanish and in AAPS selected English

Thanks again

MilosKozak, 2023-06-12:
It would help a lot to find a broken plugin.
Do not open all preferences but only preferences for a plugin to find it

jbm76, 2023-06-12:
In my case, when I had the error, I went to the config builder and I changed the algorithm, to try to navigate through the gear to the preferences, and this was also impossible to do.

Thanks again

> It would help a lot to find a broken plugin. Do not open all preferences but only preferences for a plugin to find it

fjpezuela, 2023-06-12:
> It would help a lot to find a broken plugin. Do not open all preferences but only preferences for a plugin to find it

Hello Milos,

I have been talking to a friend who also has the problem and she has been trying all the options and it doesn't work for her either. Here are all the ones she has tried:  

In "Config Builder"--> BYODA, EOPatch2, Sensivity Ofer1, OpenAPS AMA, Loop, NSClientV3, Overview, Maintenance, SMS Communicator, etc.

It does not open any configuration options.

I have videos of all the tests, but they take up more than 10 MB and it won't let me upload them, but I'll confirm that this is the case. I also tried most of the options and they didn't open either.

Thanks.

adrianagomezdiaz, 2023-06-12:
https://github.com/nightscout/AndroidAPS/assets/47492569/b46abe45-fc94-4275-a18c-11482b33fcac

[image]

Preferences don't open!!!

fjpezuela, 2023-06-12:
@MilosKozak I just found out that it happens right after the end of a patch activation. Everything was working fine for me, I just changed the patch and opened the EOPAtch2 preferences and it failed.

Attached is a screenshot and log
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/11728624/AndroidAPS.log)
[image]

fjpezuela, 2023-06-13:
Hello,

This morning I had a problem with the patch I put on last night and was able to draw the following conclusions:

When the patch is deactivated, the problem when opening the preferences or any option of any related section, is solved.

When the patch activation process is finished, you cannot open again any of the options related to the preferences.

I have recorded a video of the whole process, in case it helps you.

https://drive.google.com/file/d/18KEIjDCHZvoH_0pe72HexeEsRpCxSVCU/view?usp=drive_link

Thank you very much

MilosKozak, 2023-06-13:
i tested pairing EOPATCH with latest dev. and no issues. are you on latest dev?

jbm76, 2023-06-13:
The error of not opening the preferences from any part of AAPS happened to me with f1118e2) just before the COMMIT made today.

> i tested pairing EOPATCH with latest dev. and no issues. are you on latest dev?

fjpezuela, 2023-06-13:
> no. it was watch comm related. then PM me your config and master pass to be able replicate it

I can't send you a PM through GitHub. I'm writing you on Discord

jbm76, 2023-06-14:
In case it helps you @MilosKozak 

Today after my patch change and with the error present again, I selected the plugin of a different pump (insight, virtual, etc) and the access to the preferences worked correctly again.  then select EoPatch and it stopped working again.  

to review, in case it may be related to the bad behavior of the EOpatch plugin, in previous days I opened a ticket, #2552 , where indicated  how with the patch already deactivated, AAPS continued to show and seems count basal insulin rates.

as always, thank you very much for everything

> no. it was watch comm related. then PM me your config and master pass to be able replicate it

fjpezuela, 2023-06-14:
Hello,

I confirm what @jbm76 says. I now have the problem when opening the preferences. If I switch from EOPatch2 to Virtual Pump, I can access without problems, and when I go back to EOPatch2 it stops opening them, so it seems that the EOPatch2 pluging is the one with some problem.

I put a video where you can see the whole process: https://drive.google.com/file/d/1pdUtdMM0h47aYV_ogeXBOZfdQF0ZV_L9/view?usp=drive_link

Thanks!

fjpezuela, 2023-06-14:
Compiling the new changes and updating the version, even if they are minor changes, temporarily fixes the problem. I just installed the changes today and after the update the preferences open correctly.

MilosKozak, 2023-06-14:
temporary fixed by https://github.com/nightscout/AndroidAPS/commit/a5774b6e2eb398311c80e9e00c01bede82978633

closed 2023-06-14 by MilosKozak