nightscout/AndroidAPS #2575: Allow usage of biometric protection with PIN instead of master password

Jackenmen, 2023-06-07:
I would really want to be able to use a short PIN instead of a regular password as a protection for my settings/boluses but right now this is not really possible. I *could* set the password to only consist of numbers but that still results in a subpar experience where I get a prompt with a full keyboard that's hard to type on. On top of that, it means that I have to share this simple password for my backups as well, although this *could* be solved by me changing the password every time I want to do a backup (but it does make it less user-friendly). Therefore, I would like to request that you consider allowing combining the biometric protection with the PIN protection instead of the master password.

closed 2023-08-03 by MilosKozak

MilosKozak, 2023-08-04:
that's against the logic .....
there are some possibilities: biometric, password, PIN .... and master password is kind of recovery when common method fails

Jackenmen, 2023-08-04:
How is it against the logic... It's common for apps to allow biometric authentication with PIN as a fallback (Android does it itself, bank apps do). I'm not even saying that the master password can't be the "final" fallback if you think that makes sense (it doesn't seem to be the case for the PIN and custom password *right now* so I suggested an option to replace the fallback rather than having two forms of fallback at the same time), I just want to be able to use the PIN instead of biometrics whenever I can't scan my finger (wet hands or just whenever it's one of those times that the phone just refuses to accept your finger, probably due to wrong position) because typing in the master password is simply not practical.