nightscout/AndroidAPS #2583: Autotune avoid straying too far from source profile

cascer1, 2023-06-14:
Hi,

When I was using openAPS six years ago, I think autotune was built in such a way that it didn't deviate more than 30% from the source profile. Is a similar feature available on AndroidAPS?

I think this is useful because in my experience Autotune can start creating a kind of echo chamber for itself where some adjustments are taken to extremes; especially in situations where I usually have carbs on board (like after breakfast).

Thanks

cascer1, 2023-06-15:
I don't think that's working how it's intended. When ran as automation, it'll use Tuned as input and output profile so won't deviate more than x% from that input. But then the output is used as new 100% baseline the next time and so it can still deviate more than the configured percentage from my actual base profile.

Philoul, 2023-08-17:
It works exactly the same way OpenAPS... (Same dataset provides same results, up to 3 digit after comma, with same behaviour you notice on a few basal rates...).

To understand the reason, you have to know that all data are categorized in 3 groups at the end of autotune prep:
- datas for basal tuning (mainly nights, end of the afternoon), no COB, little deviations
- datas for Carb absorption tuning (during seral hours after each meal)
- datas for ISF tuning (a few periods with high deviation not categorized as UAM).
(the latest category of datas, when UAM is detected, datas are finaly included in previous groups for Autotune Core module.

The Min Max ratio is applied comparing the input profile only (not the intermediate calculated profile).

Concerning basal tuning, you can have a lot of periods during a day without any data to be able to calculate basal adjustments (all period with data for COB or ISF tuning...) and autotune will adjust rates by an linear interpolation of calculated basal rates.
- The Min Max ratio is not applied by OpenAPS algo (in Autotune Core module) for these interpolated values.
- Ratio is applied as limitation only for basal rates tuned by basal data.

I don't think it's very important because these periods are always during meal absorption (so with most of the case a 0 TBR running after a meal bolus).

I didn't change this during adaptation of autotune algo as AAPS plugin for 2 reasons:
- I consider OpenAPS algo as "Validated" "As is"
- I provided several dataset (input/output) calculated by OpenAPS algo for unit test of AAPS plugin to be able to check that AAPS plugin provides exactly the same results...

cascer1, 2023-08-17:
@Philoul Yes, the algorithm produces the same result given the same inputs. However, that's not what I'm trying to say here.

On OpenAPS (running on a raspberry pi when I was using a minimed 774 pump), my experience was that it would never deviate more than x% (based on autosens settings I believe) from the profile I had configured in my pump. It did not update the pump profile.

From the [documentation](https://openaps.readthedocs.io/en/latest/docs/Customize-Iterate/autotune.html#how-autotune-works):

> Autotune limits how far it can adjust (or recommend adjustment, if running autotune outside oref0 closed loop) basal, or ISF or CSF, from what is in the existing pump profile. Autotune uses the same autosens_max and autosens_min multipliers found in your preferences.json for oref0. So if autotune is running as part of your loop, autotune can’t get too far off without a chance for a human to review the changes. 

However, in AndroidAPS my pump profile is automatically updated based on the autotune output. This then shifts the reference window for the min/max deviation to that new result. In time, that adds up to a bigger deviation than I would expect given the initial profile and max deviation settings.

Philoul, 2023-08-17:
Not sure to understand your answer.

Aaps plugin works the same way than openAPS:

If you tune for ex with 5 days of data

You starts tuning with PumpProfile to tune day1 and AT will compare with PumpProfile to limit min and max values (but only for "basal periods", limitation is not applied for interpolated values of basal during cob or isf periods)

Then autotune will use day1 result to tune with day 2 data. and AT will compare day 2 result with PumpProfile (again limitations will only be applied on "basal periods").

day 2 will be used to calculate day 3 and compare will always be done with PumpProfile to limit values

Up to latest selected day...

PumpProfile is automatically updated only if 
- Autotune is launched with automation rule 
- AND in Autotune parameters,  Automatically update input profile is set to Enable...

Never set to Enable "Automatic Update" if you have important modification during you different run of Autotune...

If you run autotune manually, you can decide to update, ignore results, create a new Tuned profile to update it... And it's only (after a few days and verification that all tuned profile remain very close to PumpProfile), that you can start to use automation and automatic update...

cascer1, 2023-08-22:
Yes, the AAPS plugin works the same as OpenAPS. however, I don't think I managed to make my point clear.

OpenAPS, in my experience, does **not** update the pump profile. So, the reference profile stays the same.

AAPS **does** update the pump profile. So, the next run will have a slightly different reference profile.

for a few runs this isn't an issue, but over time those deviations add up where it can keep going to further extremes because the safety feature of only adjusting x% is now based on its previous result rather than the original profile.

Philoul, 2023-08-26:
I never used openAPS myself... But I know there is an option in OAPS to automatically run autotune daily...
How are managed profiles if you never update input profile???

Within AAPS. input profile is only updated if you request it within Autotune preferences, so you can also disable this update...