nightscout/AndroidAPS #2585: Improve pre-defined settings on watch calculator

olorinmaia, 2023-06-14:
We and school assistant are using AAPS watch to bolus for meals. We are using all features for the calculator ON to simplify bolusing for them. That way SMB's and IOB and COB will always be taken into calculation for bolus.

On rare occassion when they are bolusing while BG is rising fast, the TREND is positive and producing a high amount of insulin that really isnt necessarry to account for. Alltho negative trend is almost always useful again, therefor trend is ON but they need to be careful.

I would love the option for more configuration on the watch calculator preset settings, similar to what is possible on the quickwizard. Especially choosing trend that is positive, negative, all or OFF. 

Screenshot of settings today:
[image]

Screenshot of quick-wiz:
[image]

MilosKozak, 2023-06-14:
this is very specific usecase. I'm trying to simplify the setting of AAPS but this would add even more complexity. 
If trend is causing troubles, don't use it at all. Wizard will remember your setting

MilosKozak, 2023-06-14:
sorry

closed 2023-06-14 by MilosKozak

olorinmaia, 2023-06-14:
@MilosKozak Just to make sure you understand my issue. This is about adding more options to the wear app calculator settings in AAPS. Not changing main AAPS phone calculator. Alltho I understand and respect your reasoning if you understood it correctly and still mean this will overly complicate AAPS settings.

Since you can't toggle directly on wear app without adjusting settings on AAPS phone, toggling trend is therefore not very convenient. I agree that it is possible to turn it completely off, but I was hoping since wear app is configurable in AAPS and not toggleable on watch that similar option like in the quick wizard could be added to wear app calculator settings.