nightscout/AndroidAPS #2586: No BG values are displayed when Dexcom app shows "low"

OFlannigan, 2023-06-14:
I went hypo a few days ago (June 9th, just aroung 1:00 am). So low, that the Dexcom app was only displaying "low". But was still receiving readings, so the sensor was connected all the time.
[image]

However, during the time of "low" my AAPS was not displaying any values from the Dexcom app. You can tell by the pale-red dots in the overview, that the values are recalculated.
[image]

The BYODA-tab also shows no readings from Dexcom during the period 1:16 to 1:36.
[image]

As soon as my levels started to rise and Dexcom was displaying an actual value, rather than "low", AAPS also picked up a value and was displaying as normal.

As far as I understand this extract from the logs, the value was received in AAPS, but still not displayed. The logs also state, that the value was submitted to Nightscout, but it also can't be seen there. I have the same gap in Nightscout, that I have in AAPS.
<div class="snippet-clipboard-content notranslate position-relative overflow-auto"><pre class="notranslate"><code class="notranslate">01:21:28.458 [main] D/DATABASE: [DataReceiver.onReceive():36]: onReceive com.dexcom.g7.EXTERNAL_BROADCAST Bundle{ glucoseValues =&gt; Bundle[mParcelledData.dataSize=152]; }Bundle
01:21:28.502 [RxCachedThreadScheduler-42] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():59]: Firing EventNewBg GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN)
01:21:28.503 [RxCachedThreadScheduler-42] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():64]: Firing EventNewHistoryData 1686266486000 GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN)
01:21:28.506 [RxCachedThreadScheduler-44] D/NSCLIENT: [NSClientPlugin$onStart$5.accept():113]: ● QUEUE Resend started: NEW_DATA
01:21:28.507 [DefaultDispatcher-worker-3] D/DATABASE: [DexcomPlugin$DexcomWorker.doWorkAndLog():186]: Inserted bg GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN)
01:21:28.507 [DefaultDispatcher-worker-3] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class info.nightscout.source.DexcomPlugin$DexcomWorker
01:21:28.517 [DefaultDispatcher-worker-2] I/NSCLIENT: [DataSyncSelectorV1Impl.processChangedGlucoseValues():370]: Loading GlucoseValue data Start: 194725 GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=null, timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN) forID: 194726
01:21:28.519 [RxCachedThreadScheduler-44] D/NSCLIENT: [NSClientPlugin$onStart$5.accept():113]: ► ADD entries Sent PairGlucoseValue {"device":"AAPS-Dexcom","date":1686266486000,"dateString":"2023-06-08T23:21:26.000Z","isValid":true,"sgv":37,"direction":"Flat","type":"sgv"} 194725/194726
</code></pre><div class="zeroclipboard-container position-absolute right-0 top-0">
    <clipboard-copy aria-label="Copy" class="ClipboardButton btn js-clipboard-copy m-2 p-0 tooltipped-no-delay" data-copy-feedback="Copied!" data-tooltip-direction="w" value="01:21:28.458 [main] D/DATABASE: [DataReceiver.onReceive():36]: onReceive com.dexcom.g7.EXTERNAL_BROADCAST Bundle{ glucoseValues => Bundle[mParcelledData.dataSize=152]; }Bundle
01:21:28.502 [RxCachedThreadScheduler-42] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():59]: Firing EventNewBg GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN)
01:21:28.503 [RxCachedThreadScheduler-42] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():64]: Firing EventNewHistoryData 1686266486000 GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN)
01:21:28.506 [RxCachedThreadScheduler-44] D/NSCLIENT: [NSClientPlugin$onStart$5.accept():113]: ● QUEUE Resend started: NEW_DATA
01:21:28.507 [DefaultDispatcher-worker-3] D/DATABASE: [DexcomPlugin$DexcomWorker.doWorkAndLog():186]: Inserted bg GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN)
01:21:28.507 [DefaultDispatcher-worker-3] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class info.nightscout.source.DexcomPlugin$DexcomWorker
01:21:28.517 [DefaultDispatcher-worker-2] I/NSCLIENT: [DataSyncSelectorV1Impl.processChangedGlucoseValues():370]: Loading GlucoseValue data Start: 194725 GlucoseValue(id=194726, version=0, dateCreated=1686266488501, isValid=true, referenceId=null, interfaceIDs_backing=null, timestamp=1686266486000, utcOffset=7200000, raw=null, value=37.0, trendArrow=FLAT, noise=null, sourceSensor=DEXCOM_NATIVE_UNKNOWN) forID: 194726
01:21:28.519 [RxCachedThreadScheduler-44] D/NSCLIENT: [NSClientPlugin$onStart$5.accept():113]: ► ADD entries Sent PairGlucoseValue {&quot;device&quot;:&quot;AAPS-Dexcom&quot;,&quot;date&quot;:1686266486000,&quot;dateString&quot;:&quot;2023-06-08T23:21:26.000Z&quot;,&quot;isValid&quot;:true,&quot;sgv&quot;:37,&quot;direction&quot;:&quot;Flat&quot;,&quot;type&quot;:&quot;sgv&quot;} 194725/194726" tabindex="0" role="button" style="display: inherit;">
      <svg aria-hidden="true" height="16" viewBox="0 0 16 16" version="1.1" width="16" data-view-component="true" class="octicon octicon-copy js-clipboard-copy-icon m-2">
    <path d="M0 6.75C0 5.784.784 5 1.75 5h1.5a.75.75 0 0 1 0 1.5h-1.5a.25.25 0 0 0-.25.25v7.5c0 .138.112.25.25.25h7.5a.25.25 0 0 0 .25-.25v-1.5a.75.75 0 0 1 1.5 0v1.5A1.75 1.75 0 0 1 9.25 16h-7.5A1.75 1.75 0 0 1 0 14.25Z"></path><path d="M5 1.75C5 .784 5.784 0 6.75 0h7.5C15.216 0 16 .784 16 1.75v7.5A1.75 1.75 0 0 1 14.25 11h-7.5A1.75 1.75 0 0 1 5 9.25Zm1.75-.25a.25.25 0 0 0-.25.25v7.5c0 .138.112.25.25.25h7.5a.25.25 0 0 0 .25-.25v-7.5a.25.25 0 0 0-.25-.25Z"></path>
</svg>
      <svg aria-hidden="true" height="16" viewBox="0 0 16 16" version="1.1" width="16" data-view-component="true" class="octicon octicon-check js-clipboard-check-icon color-fg-success m-2 d-none">
    <path d="M13.78 4.22a.75.75 0 0 1 0 1.06l-7.25 7.25a.75.75 0 0 1-1.06 0L2.22 9.28a.751.751 0 0 1 .018-1.042.751.751 0 0 1 1.042-.018L6 10.94l6.72-6.72a.75.75 0 0 1 1.06 0Z"></path>
</svg>
    </clipboard-copy>
  </div></div>

Here's an extract from Nightscout as well, showing the same gap as AAPS.
[image]

Running the patched Dexcom app's latest release 1.6 and AAPS 3.2.0 dev i.

Here are the full logs for that specific period:
[AndroidAPS._2023-06-09_00-45-52_.0.zip](https://github.com/nightscout/AndroidAPS/files/11749374/AndroidAPS._2023-06-09_00-45-52_.0.zip)
[AndroidAPS._2023-06-09_00-45-52_.1.zip](https://github.com/nightscout/AndroidAPS/files/11749375/AndroidAPS._2023-06-09_00-45-52_.1.zip)

MilosKozak, 2023-06-15:
low is not BG but error value (37 in numbers)

OFlannigan, 2023-06-15:
Thanks for your comment.
I'm a bit confused actually. I've just recent made the change to G7 and was using G6 wiht BYODA before.
Luckily, I wasn't that low very often but when I was, I still had readings in AAPS (40 or sometimes even below).

Just out of curiosity: Is the 37 as error value definded by the Dexcom app or by AAPS?

MilosKozak, 2023-06-15:
dexcom
xdrip shows this as an invalid vaule too

closed 2023-08-03 by MilosKozak