nightscout/AndroidAPS #2587: Compiling Dev branch

JoanoToxo, 2023-06-14:
Hi, I am compiling with android studio the dev branch of androidaps.
fullRelease variant is ok
nsClientRelease i have an error:
Task 'assembleNsclientRelease' not found in project ':app'. Some candidates are: 'assembleAapsclientRelease'.

* Try:
> Run gradle tasks to get a list of available tasks.
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.

My versions programs:
Android Studio Flamingo | 2022.2.1 Patch 2
Build #AI-222.4459.24.2221.10121639, built on May 12, 2023
Runtime version: 17.0.6+0-b2043.56-9586694 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Windows 7 6.1
GC: G1 Young Generation, G1 Old Generation
Memory: 1280M
Cores: 4
Registry:
    external.system.auto.import.disabled=true
    ide.text.editor.with.preview.show.floating.toolbar=false
    gradle.version.catalogs.dynamic.support=true

Non-Bundled Plugins:
    com.intellij.marketplace (222.4459.28)
    com.thoughtworks.gauge (222.4459.16)

thks

JoanoToxo, 2023-06-14:
[image]
[image]

JoanoToxo, 2023-06-15:
[image]
In the "Generate Signed Bundle or APS", at the Build Variants" I haven't the aapsclientRelease variant

parapenT1sta, 2023-06-15:
[image]

JoanoToxo, 2023-06-15:
dev branch:
[image]
[image]
[image]
I don't see the same variants as you!!!

parapenT1sta, 2023-06-15:
Did you create a new project with `https://github.com/nightscout/AndroidAPS.git`

JoanoToxo, 2023-06-15:
Yes. I have created s new project.
I have deleted the android studio and install again.

parapenT1sta, 2023-06-15:
Delete the StudioProjects folders that you have.

JoanoToxo, 2023-06-15:
I've already done it. delete the StudioProjects directory, uninstall android studio and go back and do the installation as indicated in the official documentation

MilosKozak, 2023-06-15:
do 
clone
switch to dev
sync project
[image]
the aapsclient appear

JoanoToxo, 2023-06-16:
With sync project the aapsclient appear
Compile ok