nightscout/AndroidAPS #2599: 3.2.0-dev-j app hangs (until app closed and restarted)

sdolenc, 2023-06-22:
We recently switched from 3.1.0.3 to 3.2.0-dev-j (commit https://github.com/nightscout/AndroidAPS/commit/a5774b6e2eb398311c80e9e00c01bede82978633 on 2023-06-14) on our pixel a4 running android 13.

We noticed the dev branch "hangs" periodically for us. The OS asks if we want to "wait" or "close" the app. We eventually need to close and restart

Example from logs:
hang at 19:57:18.630
app restart at 20:03:42.198
[short log file](https://github.com/nightscout/AndroidAPS/files/11835487/AndroidAPS._2023-06-21_20-04-12_.17.small.log)
[full log file zipped](https://github.com/nightscout/AndroidAPS/files/11835496/AndroidAPS._2023-06-21_20-04-12_.17.zip)

I'm not sure if this is related or a separate concern: we also get more omnipod pump errors/alerts on the dev branch. Here's a screenshot of the four we got this morning.
[screenshot](https://github.com/nightscout/AndroidAPS/assets/21182133/5ffa36b1-7933-4874-b314-03897b8fbabf)

jbr7rr, 2023-06-24:
I had a quick look for the pod errors, it seems that your connection is very unstable, can you provide a log file with also with the errors "Bolus delivery status uncertain", "Failed to resume delivery"? At this point I'm not sure it is related to the app hanging.

sdolenc, 2023-06-24:
Sorry I didn't grab those logs. We ended up replacing the pod and moved back to 3.1.0.3

I'm going to close this since I don't think there's any further action that can be taken. Maybe it can be reactivated if another person experiences similar issues

closed 2023-06-24 by sdolenc