nightscout/AndroidAPS #2602: [Latest DEV] No BG upload to NS when BG Source is xDrip

parapenT1sta, 2023-06-25:
Nightscout: Latest DEV Version 15
xDrip source: Dexcom G5
AAPS: Latest DEV
Logs (check around 18:53:46):  
[AndroidAPS_LOG_1687715846423.log.zip](https://github.com/nightscout/AndroidAPS/files/11860468/AndroidAPS_LOG_1687715846423.log.zip)

Now that xDrip can received all data from AAPS, I disabled NS upload and download in xDrip and I had set all upload to NS from AAPS but I can check that BG value isn't shown in NS, as screenshot below. As per NSV3 the BG value is sent.

NS should had received BG 90 but got stuck with 88. I don't know if had worked before because I always had xDrip to upload and download from NS. I would like to have AAPS dealing with this only.

[image]

[image]

[image]

[image]

ct2iqj, 2023-06-25:
I have the same issue, xDrip as source for Libre 2. As you can check OpenAPS information was updated 1 minute ago but still no BG readings for 37 minutes.

jimbaxter, 2023-07-14:
I had the same problem, the only mesage I have is "Started AndroidAPS" and then no data.

moamo67, 2023-07-30:
Same problème no gly uploaded from aaps nsv3

MilosKozak, 2023-08-21:
resolved in dev-k

closed 2023-08-21 by MilosKozak

szantos, 2023-08-24:
Issue still present with AAPS 719192e

NS dev 15: d602de1; xDrip-variant2: 2023-08-19

The long story:
I've used Dexcom G6 with BYODA for more than 1,5 years. AAPS did all uploading, since 2023-04-01 with v3 API. During the last weeks BYODA started to lose BT connection - don't know why. The phone is a Huawei P30 Pro, last update is from December 2022, no updates to BYODA lately. I have tried an older BYODA version and another transmitter, too - similar problems after some days. Therefore I've switched to xDrip (variant2) to handle the CGM. Reason for variant2 is, that I've already had an xDrip instance for alerts and widget as an AAPS-follower, which I did not wanted to reconfigure.

CGM is now in xDrip-variant2 (instead of BYODA), and AAPS stopped uploading SGV, only treatments - at least I don't see any entries in NS.

I've tried to reconfigure AAPS to upload anything but SGV, and let xDrip-variant2 do the uploading of SGV via v1 API. Now I see only SGV in NS, but no treatments.

I would appreciate any suggestions.

MilosKozak, 2023-08-24:
open new issue with screenshots and logs