nightscout/AndroidAPS #2603: [Master] Nightacout crash caused a permanent local db corruption?

ninelore, 2023-06-28:
Both nightacout and aaps master versions

It started when i was noticing that NS didnt recieve anything anymore. For a weird reason i was getting socketexceptiona in ns client.
Restarting aaps wpuldnt help, ao i redeployed nightacout from newest master. Problem was solved for a time. 

A day later my aaps would not calculate anything anymore. A loop run was lasting minutes and no cob or iob would be calculated at all. Aaps restarts and reinstalls wouldnt solve the issue, only a DB reset did.

ninelore, 2023-06-28:
Logs will follow

MilosKozak, 2023-08-03:
where are the logs?

ninelore, 2023-08-03:
Sorry forgot to hit send.

I was unable to get Logs of the time when it happend. Feel free to close as not reproducable or stale.

closed 2023-08-04 by MilosKozak