nightscout/AndroidAPS #2607: AAPS doesn't run on new BG latest dev-j

yhonatanb20, 2023-07-01:
AAPS doesn't always run on new BG, which causes it to actually get working sometimes only every 10 minutes instead of 5. That is, BG reading is recieved all right but for some reason the loop just doesn't run itself. It happens sporadically with no apparent reason, in all AAPS versions since 3.0.0 as far as I can remember, just now it happens very frequently (many times a day)
All play protect and battery optimization whatsoever are off.
[AndroidAPS._2023-06-28_03-00-31_.0.zip](https://github.com/nightscout/AndroidAPS/files/11927179/AndroidAPS._2023-06-28_03-00-31_.0.zip)
[AndroidAPS._2023-06-28_03-00-31_.1.zip](https://github.com/nightscout/AndroidAPS/files/11927178/AndroidAPS._2023-06-28_03-00-31_.1.zip)
[AndroidAPS._2023-06-28_12-07-57_.2.zip](https://github.com/nightscout/AndroidAPS/files/11927177/AndroidAPS._2023-06-28_12-07-57_.2.zip)

[AndroidAPS._2023-06-28_12-07-57_.3.zip](https://github.com/nightscout/AndroidAPS/files/11927176/AndroidAPS._2023-06-28_12-07-57_.3.zip)

Phone model: Samsung Galaxy A52s
Android version: Android 13 One UI 5.1
AAPS version: 3.2.0 dev-j
Build: 7920141834-2023.06.30
BG data source: Dexcom G6 BYODA
Time: i.e. 00:07
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/11927145/AndroidAPS.log)
[AndroidAPS._2023-07-01_16-54-51_.4.zip](https://github.com/nightscout/AndroidAPS/files/11927147/AndroidAPS._2023-07-01_16-54-51_.4.zip)
[AndroidAPS._2023-07-01_16-54-51_.3.zip](https://github.com/nightscout/AndroidAPS/files/11927148/AndroidAPS._2023-07-01_16-54-51_.3.zip)
[AndroidAPS._2023-07-01_03-00-10_.2.zip](https://github.com/nightscout/AndroidAPS/files/11927149/AndroidAPS._2023-07-01_03-00-10_.2.zip)
[AndroidAPS._2023-07-01_03-00-10_.1.zip](https://github.com/nightscout/AndroidAPS/files/11927150/AndroidAPS._2023-07-01_03-00-10_.1.zip)
[AndroidAPS._2023-07-01_03-00-10_.0.zip](https://github.com/nightscout/AndroidAPS/files/11927151/AndroidAPS._2023-07-01_03-00-10_.0.zip)
[AndroidAPS._2023-06-30_23-20-42_.7.zip](https://github.com/nightscout/AndroidAPS/files/11927152/AndroidAPS._2023-06-30_23-20-42_.7.zip)
[AndroidAPS._2023-06-30_23-20-42_.6.zip](https://github.com/nightscout/AndroidAPS/files/11927153/AndroidAPS._2023-06-30_23-20-42_.6.zip)
[AndroidAPS._2023-06-30_17-58-14_.5.zip](https://github.com/nightscout/AndroidAPS/files/11927154/AndroidAPS._2023-06-30_17-58-14_.5.zip)
[AndroidAPS._2023-06-30_17-58-14_.4.zip](https://github.com/nightscout/AndroidAPS/files/11927155/AndroidAPS._2023-06-30_17-58-14_.4.zip)
[AndroidAPS._2023-06-30_03-00-32_.3.zip](https://github.com/nightscout/AndroidAPS/files/11927156/AndroidAPS._2023-06-30_03-00-32_.3.zip)
[AndroidAPS._2023-06-30_03-00-32_.2.zip](https://github.com/nightscout/AndroidAPS/files/11927157/AndroidAPS._2023-06-30_03-00-32_.2.zip)
[AndroidAPS._2023-06-30_03-00-32_.1.zip](https://github.com/nightscout/AndroidAPS/files/11927158/AndroidAPS._2023-06-30_03-00-32_.1.zip)
[AndroidAPS._2023-06-30_03-00-32_.0.zip](https://github.com/nightscout/AndroidAPS/files/11927159/AndroidAPS._2023-06-30_03-00-32_.0.zip)
[AndroidAPS._2023-06-29_06-33-08_.7.zip](https://github.com/nightscout/AndroidAPS/files/11927160/AndroidAPS._2023-06-29_06-33-08_.7.zip)
[AndroidAPS._2023-06-29_06-33-08_.6.zip](https://github.com/nightscout/AndroidAPS/files/11927161/AndroidAPS._2023-06-29_06-33-08_.6.zip)
[AndroidAPS._2023-06-29_06-33-08_.5.zip](https://github.com/nightscout/AndroidAPS/files/11927162/AndroidAPS._2023-06-29_06-33-08_.5.zip)
[AndroidAPS._2023-06-29_06-33-08_.4.zip](https://github.com/nightscout/AndroidAPS/files/11927163/AndroidAPS._2023-06-29_06-33-08_.4.zip)
[AndroidAPS._2023-06-29_06-33-08_.3.zip](https://github.com/nightscout/AndroidAPS/files/11927164/AndroidAPS._2023-06-29_06-33-08_.3.zip)
[AndroidAPS._2023-06-29_06-33-08_.2.zip](https://github.com/nightscout/AndroidAPS/files/11927165/AndroidAPS._2023-06-29_06-33-08_.2.zip)
[AndroidAPS._2023-06-29_06-33-08_.1.zip](https://github.com/nightscout/AndroidAPS/files/11927169/AndroidAPS._2023-06-29_06-33-08_.1.zip)
[AndroidAPS._2023-06-29_06-06-43_.0.zip](https://github.com/nightscout/AndroidAPS/files/11927170/AndroidAPS._2023-06-29_06-06-43_.0.zip)
[AndroidAPS._2023-06-28_12-07-57_.7.zip](https://github.com/nightscout/AndroidAPS/files/11927171/AndroidAPS._2023-06-28_12-07-57_.7.zip)
[AndroidAPS._2023-06-28_12-07-57_.6.zip](https://github.com/nightscout/AndroidAPS/files/11927172/AndroidAPS._2023-06-28_12-07-57_.6.zip)
[AndroidAPS._2023-06-28_12-07-57_.5.zip](https://github.com/nightscout/AndroidAPS/files/11927173/AndroidAPS._2023-06-28_12-07-57_.5.zip)
[AndroidAPS._2023-06-28_12-07-57_.4.zip](https://github.com/nightscout/AndroidAPS/files/11927175/AndroidAPS._2023-06-28_12-07-57_.4.zip)

yhonatanb20, 2023-07-02:
Here is another example. BG recieved at 20:56 but aaps didn't invoke
[image]
[image]
[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/11930357/AndroidAPS.log)
[AndroidAPS._2023-07-02_03-00-21_.4.zip](https://github.com/nightscout/AndroidAPS/files/11930358/AndroidAPS._2023-07-02_03-00-21_.4.zip)
[AndroidAPS._2023-07-02_03-00-21_.3.zip](https://github.com/nightscout/AndroidAPS/files/11930359/AndroidAPS._2023-07-02_03-00-21_.3.zip)
[AndroidAPS._2023-07-02_03-00-21_.2.zip](https://github.com/nightscout/AndroidAPS/files/11930360/AndroidAPS._2023-07-02_03-00-21_.2.zip)
[AndroidAPS._2023-07-02_03-00-21_.1.zip](https://github.com/nightscout/AndroidAPS/files/11930361/AndroidAPS._2023-07-02_03-00-21_.1.zip)
[AndroidAPS._2023-07-02_03-00-21_.0.zip](https://github.com/nightscout/AndroidAPS/files/11930362/AndroidAPS._2023-07-02_03-00-21_.0.zip)
[AndroidAPS._2023-07-01_16-54-51_.6.zip](https://github.com/nightscout/AndroidAPS/files/11930363/AndroidAPS._2023-07-01_16-54-51_.6.zip)
[AndroidAPS._2023-07-01_16-54-51_.5.zip](https://github.com/nightscout/AndroidAPS/files/11930364/AndroidAPS._2023-07-01_16-54-51_.5.zip)
[AndroidAPS._2023-07-01_16-54-51_.4.zip](https://github.com/nightscout/AndroidAPS/files/11930365/AndroidAPS._2023-07-01_16-54-51_.4.zip)
[AndroidAPS._2023-07-01_16-54-51_.3.zip](https://github.com/nightscout/AndroidAPS/files/11930366/AndroidAPS._2023-07-01_16-54-51_.3.zip)
[AndroidAPS._2023-07-01_03-00-10_.2.zip](https://github.com/nightscout/AndroidAPS/files/11930367/AndroidAPS._2023-07-01_03-00-10_.2.zip)
[AndroidAPS._2023-07-01_03-00-10_.1.zip](https://github.com/nightscout/AndroidAPS/files/11930368/AndroidAPS._2023-07-01_03-00-10_.1.zip)
[AndroidAPS._2023-07-01_03-00-10_.0.zip](https://github.com/nightscout/AndroidAPS/files/11930369/AndroidAPS._2023-07-01_03-00-10_.0.zip)
[AndroidAPS._2023-06-30_23-20-42_.7.zip](https://github.com/nightscout/AndroidAPS/files/11930370/AndroidAPS._2023-06-30_23-20-42_.7.zip)
[AndroidAPS._2023-06-30_23-20-42_.6.zip](https://github.com/nightscout/AndroidAPS/files/11930371/AndroidAPS._2023-06-30_23-20-42_.6.zip)
[AndroidAPS._2023-06-30_17-58-14_.5.zip](https://github.com/nightscout/AndroidAPS/files/11930372/AndroidAPS._2023-06-30_17-58-14_.5.zip)
[AndroidAPS._2023-06-30_17-58-14_.4.zip](https://github.com/nightscout/AndroidAPS/files/11930373/AndroidAPS._2023-06-30_17-58-14_.4.zip)
[AndroidAPS._2023-06-30_03-00-32_.3.zip](https://github.com/nightscout/AndroidAPS/files/11930374/AndroidAPS._2023-06-30_03-00-32_.3.zip)
[AndroidAPS._2023-06-30_03-00-32_.2.zip](https://github.com/nightscout/AndroidAPS/files/11930375/AndroidAPS._2023-06-30_03-00-32_.2.zip)
[AndroidAPS._2023-06-30_03-00-32_.1.zip](https://github.com/nightscout/AndroidAPS/files/11930376/AndroidAPS._2023-06-30_03-00-32_.1.zip)
[AndroidAPS._2023-06-30_03-00-32_.0.zip](https://github.com/nightscout/AndroidAPS/files/11930377/AndroidAPS._2023-06-30_03-00-32_.0.zip)
[AndroidAPS._2023-06-29_06-33-08_.7.zip](https://github.com/nightscout/AndroidAPS/files/11930378/AndroidAPS._2023-06-29_06-33-08_.7.zip)
[AndroidAPS._2023-06-29_06-33-08_.6.zip](https://github.com/nightscout/AndroidAPS/files/11930379/AndroidAPS._2023-06-29_06-33-08_.6.zip)
[AndroidAPS._2023-06-29_06-33-08_.5.zip](https://github.com/nightscout/AndroidAPS/files/11930380/AndroidAPS._2023-06-29_06-33-08_.5.zip)
[AndroidAPS._2023-06-29_06-33-08_.4.zip](https://github.com/nightscout/AndroidAPS/files/11930381/AndroidAPS._2023-06-29_06-33-08_.4.zip)
[AndroidAPS._2023-06-29_06-33-08_.3.zip](https://github.com/nightscout/AndroidAPS/files/11930382/AndroidAPS._2023-06-29_06-33-08_.3.zip)
[AndroidAPS._2023-06-29_06-33-08_.2.zip](https://github.com/nightscout/AndroidAPS/files/11930383/AndroidAPS._2023-06-29_06-33-08_.2.zip)
[AndroidAPS._2023-06-29_06-33-08_.1.zip](https://github.com/nightscout/AndroidAPS/files/11930384/AndroidAPS._2023-06-29_06-33-08_.1.zip)
[AndroidAPS._2023-06-29_06-06-43_.0.zip](https://github.com/nightscout/AndroidAPS/files/11930386/AndroidAPS._2023-06-29_06-06-43_.0.zip)
[AndroidAPS._2023-06-28_12-07-57_.7.zip](https://github.com/nightscout/AndroidAPS/files/11930387/AndroidAPS._2023-06-28_12-07-57_.7.zip)

yhonatanb20, 2023-07-18:
Why is the issue ignored? Issue still persists...

MilosKozak, 2023-07-20:
are you sure BG are received on time and not only backfilled?

yhonatanb20, 2023-07-20:
BG source set to BYODA, backfill data in set to off, logs say DEXCOM_G6_NATIVE so pretty sure that yes...
Is there anything else I should check?
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/12110229/AndroidAPS.log)

[image]
[image]
[image]
[image]

yhonatanb20, 2023-07-20:
For example here  @MilosKozak, bg data recieved at 23:37 but no firing event:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/12113895/AndroidAPS.log)

[image]
[image]
[image]
[image]
[image]

yhonatanb20, 2023-07-25:
ISSUE SOLVED!!!
Data actually was backfilled from NS due to old NS plugin. Switching to NS v3 new plugin solved the problem!

closed 2023-07-25 by yhonatanb20