nightscout/AndroidAPS #2612: AAPS only sporadically receives the BG values from XDrip. DEV-j

elgrupo123, 2023-07-05:
The BG values arrive very irregularly. Sometimes a 2 hour break. A restart sometimes helps.
At XDrip, the values from the Juggluco are stable and regular.
The path directly from Juggluco to AAPS (XDrip Broadcast) also works perfectly.
I only see the problem in the DEV version. With Master there are no problems.

vanelsberg, 2023-07-13:
Most likely cause is xDrip+ not properly running in the background? Suggesting to double check you phone's power saving options on xDrip+. 

(Tip: Maybe check out "DON'T KILL MY APP?" on the AndroidAPS about popup)

andyrozman, 2023-07-15:
What kind of phone do you have? For some phones you need to turn on certain
options in xdrip.

Andy

elgrupo123, 2023-07-15:
Pixel 5 and Pixel 7. As already written, the problem only affects the dev version, not the master version.  The sync with V3/NS 15 is also very unreliable.  With the old sync method everything works fine.

andyrozman, 2023-07-15:
Mhm. AAPS should be receiving broadcasts not using NS for receiving Bg
data. Can you look at logs and see if reading are read every 5 minutes or
if some get lost (in xdrip) ?
Did this started happening with specific dev version?

Andy

elgrupo123, 2023-07-16:
the values all arrive at XDrip without jumps.  Just not with AAPS.  According to my observation it affects the DEV j version.  Unfortunately it is not usable.  That's why we use Juggluco's XDrip Broadcast.  Works flawlessly.

[image]

vanelsberg, 2023-07-16:
**Tested on test phone:**
- xDrip+ (build e50dfdf-2023.05.03) reliably receiving BG and treatments from NS/V15
- AAPS 3.2.0-dev-j (lastest dev build e9971b5ec9-2023-07.15) configured BG source = xDrip+ & no syncing with NS

**Result:**
- xDrip+ receiving new BG values and treatments from Hardware Data Source "NS" every 5 mins
- _AAPS: No BG data receiving_, no activity on XDRIP+BG tab.

Checked several settings on xDrip+ and could not find anything wrong.
But no BG in AAPS at all. Looks like confirmation of the above?

@MilosKozak Should this "test" setup work or am I missing something?
(When yes, I can will some logging from AAPS and xDrip)

andyrozman, 2023-07-17:
I am using newer xdrip (20230608) with much older AAPS (3.2 dev from few
months ago) and broadcasting works, so maybe something changed on new dev
that is impacting receiving of broadcasts from Xdrip?

Andy

elgrupo123, 2023-07-17:
That's my guess too. The master version doesn't have the problem. Unfortunately I can't test an older dev version.

Hardware Data Source "NS" every 5 mins - *AAPS: No BG data receiving*, no activity on XDRIP+BG tab. Checked several settings on xDrip+ and could not find anything wrong. But no BG in AAPS at all. Looks like confirmation of the above? @MilosKozak <https://github.com/MilosKozak> Should this "test" setup work or am I missing something? (When yes, I can will some logging from AAPS and xDrip)

vanelsberg, 2023-07-18:
> I am using newer xdrip (20230608) with much older AAPS (3.2 dev from few months ago) and broadcasting works, so maybe something changed on new dev that is impacting receiving of broadcasts from Xdrip?

Right. I will see if I can do some testing to see if I can find what build possibly started having problems.

vanelsberg, 2023-07-18:
> I am using newer xdrip (20230608) with much older AAPS (3.2 dev from few months ago) and broadcasting works, so maybe something changed on new dev that is impacting receiving of broadcasts from Xdrip?

@andyrozman @elgrupo123 Ok, **it does works for me, also with latest dev builds.** Needed to set _Settings -> Inter-app settings -> Identify receiver_ to get it working (https://wiki.aaps.app/en/latest/Configuration/xdrip.html#xdrip-identify-receiver)
So my bad: has been long since I tested xDrip+ for BG source. Never needed to set this so overlooked that option.

Have been getting BG value for 3 hours. Will run this setup for a few days to see if I can repro the posters problem.

elgrupo123, 2023-07-20:
Unfortunately I cannot confirm that. Even reinstalling XDrip does not fix the error - for me.

vanelsberg, 2023-07-20:
> Will run this setup for a few days to see if I can repro the posters problem.

After 24 hours (phone in idle with lockscreen) not a single BG value missed. So working as expected.
Details.
Test Phone: Nokia 5.4, Android 12
- DASH Drive but no Pod activated
- Not syncing with NS
- BG source is xDrip+ (version 2023.07.16)
- xDrip+ is getting BG value's from my main NS

vanelsberg, 2023-07-20:
> Even reinstalling XDrip does not fix the problem - for me

What version of xDrip+ are you running?
@MilosKozak I tested with AAPS. Could this be a problem specific to AAPSClient?

vanelsberg, 2023-07-20:
> ...does not fix the problem - for me.

Still suspect this may be a problem with Pixel ad battery management.
Did you try disabling Google Play Protect? (it can close app's that it "thinks" are misbehaving)

andyrozman, 2023-07-20:
You might need post a question in xdrip community...
So in beginning of year I had similar problem with OnePlus (xdrip was
receiving every 2nd reading) so OnePlus was put on special list (together
with Samsung and Redmi) of phones that have "broken" power management with
BT devices. Maybe your device needs to be put on that list too?
Andy

elgrupo123, 2023-07-30:
Apparently an XDrip problem after all. With this version it runs much better, but also not perfect.
https://github.com/blaqone/xDrip

closed 2023-07-30 by elgrupo123