nightscout/AndroidAPS #2619: Suggestion: Add smoothing/filtering option for jumpy heart rate data points

vanelsberg, 2023-07-13:
**AndroidAPS 3.2.0-dev-j / Wear app**
Build/version: 37DF642C50-2023.07.12
Flavor: fullrelease
**Watch**: Samsung Galaxy Watch4 Classic, Wear OS 3.5

**Observation:**
Heart Rate measurements received by AAPS as showed on overview are quite jumpy. I think mostly because of occasional inaccurate readings due to watch movements. Samsung health shows min/max ranges and/or calculates heart rate at rest values that are quite accurate.

**Suggestion:**
For AAPS to show or act in some way it would help to do some smoothing by averaging data or some smart filtering in "sudden out-of-band data points"?

**AAPS overview sample:**

[image]

jbr7rr, 2023-07-17:
I see the same thing with my Galaxy Watch 4. Typically when my heart rate is actaully around 60, it bounces somewhere between 120 and 60. Sometimes it will even report a heart rate of 120+ overnight for a couple of hours, which is highly unlikely for me. Also I do actually see this behavior also in Samsung Health. I think for some reason the heart rate gets doubled sometimes?

In this case I don't think smoothing will fix bad sensors. And I think that heart rate based features should come with a warning that this can happen.

vanelsberg, 2023-07-21:
> I see the same thing with my Galaxy Watch 4. Typically when my heart rate is actually around 60, it bounces somewhere between 120 and 60.

Yes, seeing the same. But not continuedly? Looks to me like the obviously "doubled" values for most part are causing these spikes in the current graph?

I think the problem is that the non-intrusive measurement method used fails due to the watch moving while being active. However even as Samsung Health shows HR ranges it manages to measure/calculate the "heart rate at rest" pretty accurate for me. And when I compare the jumpy HR lines to actual HR values, when imaging an "average" line on while ignoring the spikes  to me it appears to get close to the actual HR trends.

So I can imagine that a simple averaging filter,  maybe excluding some of the obvious "out of band / doubled" values could give a much more stable HR trend result that can be used for signaling low/high activity?

vanelsberg, 2023-08-14:
Did some testing, trying to do some basic averaging of the HR data coming in. But data is so much jumpy I wasn't able to get any real improvement. So I gave up....

Assuming Samsung Health is doing some smart post-processing of the HR data to get something sensible info?

Philoul, 2024-11-23:
@vanelsberg  With this commit https://github.com/nightscout/AndroidAPS/commit/b57a9d0c0c96b72da6e68425c5e7631be55b8642 included in dev 5 or 6 month ago (Setting in Watch to smooth data with 5min, 10min or 15min of sliding average), can we close this issue/enhancement ?

vanelsberg, 2024-11-25:
> @vanelsberg With this commit [b57a9d0] ...  can we close this issue/enhancement ?

Actually forgot about this until last week when a started retesting on current dev:
Now on Galaxy Watch 6, getting rather fair values when on _15 minutes average_.

I would like to request just one small change/addition: also add a period of 30 minutes for averaging?
Reason: For the purpose of saving power I have my watch set to measure HR every 10 minutes. So averaging over a period less then 10 minutes would not really work.

With a 30-minute period averaging it should have at least 2 new HR values for every sample which I expect would result in much smoother data, better cancelling out faulty readings (that still happen)

I'll DM you on this, for you to decide on above...

MilosKozak, 2026-10-11:
The watch app has an "HR Smoothing" setting that averages heart rate over 5, 10 or 15 minutes. If you still need a 30 minute option, please open a new small request for it.

closed 2026-10-11 by MilosKozak