nightscout/AndroidAPS #2625: Manual resets needed to restore RileyLink connection

refields, 2023-07-16:
AndroidAPS version 3.1.0.3 on Cubot Pocket
Medtronic Pump
Issue reproduced with Orange Link and Ema Link
--------------
Following lost connection to RileyLink, the only way to restore connection is to simultaneously reset the RileyLink device and force stop / re-open AndroidAPS.  This has been reported by a few other users as well.

Can a routine be added in AndroidAPS to re-establish a connection with the RileyLink so that this manual reset by user is not required?

andyrozman, 2023-07-17:
No. There can be different reason why this is happening.
1. Rileylink gets stuck
2. AAPS still thinks it's connected.

So for 1. There is no solution. If RL gets stuck you just have to restart.

There would be possible solution for number 2. But it would require big
changes in whole framework Restart AAps restart should be enough in this
case.
Abdy

refields, 2023-07-31:
Thanks for responding.  Couldn't 2 be resolved with a simple statement if Last Connection is greater than some amount to attempt a connection reset, say on 10 minute intervals or something of the like?

old-square-eyes, 2023-10-06:
Recently this has been happening a lot for me. It's waking me up at night and causing disruption at work. 3.1.

I suspect it's exacerbated by other BT activity on my S22 or perhaps Samsung/Android OS. I have disabled dynamic power management as it seems to ignore my "Do not optimise" directive. Unfortunately you can no longer set the BT system app to not optimise.

Two decades in and BT is still a shit show.

old-square-eyes, 2023-10-06:
For more context. My OL, which are difficult to get hold of here in New Zealand, is wearing out having to open and close the lid multiple times a day. Appreciate that's a OL thing but anything that can be done to stablabise the connection will help.

MilosKozak, 2026-10-11:
Two bugs that stopped the RileyLink connection until AAPS was restarted are fixed (in 3.3.2.0 and in the current dev version). If the RileyLink itself hangs, it still needs a reset. Please reopen with logs if you still see this on a new version.

closed 2026-10-11 by MilosKozak