nightscout/AndroidAPS #2626: [dev-j] AAPS or AAPSClient pushing multiple profile database records to NS

kristinsmotz, 2023-07-17:
After I updated to dev-j on 16th June, it seems there is a bug where with every loop run (or even more frequently), either AAPS or AAPSClient or both, send a new profile database record. 

I remember last year (oct-nov 22 I think) there was a bug creating multiple profile database records with the same StartDate, but now it is creating with a new value in Created AND in StartDate field. 

If I pause NSclient on both apps, it stops posting new records, but if I unpause, it's pushing new records again. 
[image]

I was able to find a script to remove the duplicate profile values for last year's problem, and customize it for this new issue to get rid of records, here you can see example of created vs startdate (valid from) for the records:
[image]

Attached logs here in case you can find some useful info there. 
[AndroidAPS_LOG_1689617704715.log.zip](https://github.com/nightscout/AndroidAPS/files/12072763/AndroidAPS_LOG_1689617704715.log.zip)

For now it seems a useful workaround to go back to dev-i version, as it's not currently pushing extra profile store records on that version.

rosseland, 2023-07-22:
I can verify this, same issue since update from dev-i to 18f01d7b5930252a01bdbda86d56a1af720a0aa0 yesterday.
@kristinsmotz Do you have your updated script avaliable? Im sure there are some of us running on dev-j :)

rosseland, 2023-07-22:
I deleted the profile nsclient reuploaded over and over. And now it seems as if the error is gone. But i ran into another bug. Im now unable to add new profiles. Ill create a new issue when i got some more info

Alamo04, 2023-07-23:
This is an old issue and was fixed a long time ago.
Can't confirm if it is back, because since NS Dev 15.0.0 I can see only my active profile.
NS Issue done in 03/2023 and no response. 

Since I changed to NS Dev 15.0.0 I can't view, change, or clone older profiles. Also can't create a new one.
This will become a big problem for parents.
Why can you see that wuth Dev-J and NS 15.0.0 ?
[image]

someguygg, 2023-07-24:
> Since I changed to NS Dev 15.0.0 I can't view, change, or clone older profiles. Also can't create a new one. This will become a big problem for parents. Why can you see that with Dev-J and NS 15.0.0 ?
> 

It doesn't show the editor in chrome. Try edge browser and you can modify NS profiles for now.

Alamo04, 2023-07-28:
@someguygg 
Chrome works fine, also Firefox. Didn't use Edge. This is a problem on NightScout Dev side.
ZehnBe fixed that for me. It was a small patch in NightScout code, on server side.

ohjay73, 2023-08-29:
I'm experiencing the same problem.. each time WS CREATE/UPDATE is triggered in aaps or aapsclient an entry is created in nightscout that takes the lowest alphabetical profile and triggers it as active with the same created and start date. Happens every 3 to 5 minutes and fills your profiles DB in mongo

kristinsmotz, 2023-08-30:
This is still a problem in 3.2 RC1, which will be a huge problem for us.

ohjay73, 2023-08-31:
> This is an old issue and was fixed a long time ago. Can't confirm if it is back, because since NS Dev 15.0.0 I can see only my active profile and unfortunately this NS Version is needed for AAPS Dev. NS Issue done in 03/2023 and no response. But as far I can see in NS Client v3, everything works ok, no spamming profile switches here.
> 
> Since I changed to NS Dev 15.0.0 I can't view, change, or clone older profiles. Also can't create a new one. This will become a big problem for parents. Why can you see that with Dev-J and NS 15.0.0 ?
> 
> [image]

Remove your authorization in nightscout and then reload the page and enter your code but DONOT check the remember me tickbox. Then check how many profiles you have.. I'm betting that by the date of creation of your profile being the same date as your comment that you are being flooded but the remember me bug is only letting you see the top level profile. Please test and report back

blochj, 2023-08-31:
I can confirm the same issue here. This fills entire mongodb database of 512mb within few hours...
[image]

MilosKozak, 2023-08-31:
something is modifying profile in NS
Do following:
change api secret (and tokens if using) to be sure nothing has write access
update api secret on main phone only
see the results
you can do one profile modification to see if it's properly updated in NS and no update loop started
then you can update api secret  on another apps (one by one with testing)

kristinsmotz, 2023-08-31:
Between the first time I saw this and updating to 3.2RC1 yesterday, I had actually switched nightscout sites and API_SECRET due to kid changing names. So there shouldn't be anything accessing the profile store except AAPS and AAPSClient apps. The problem pops up right after updating to new version (have tried going back and forth between dev-i and newer versions), even if I don't initiate a profile switch or profile edit. In the NSClient tab on AAPS phone it keeps sending new "profile store" batches every 10-ish seconds. 

Currently we have the following setup: 
FSL3 being read by Juggluco, uploading glucose values to nightscout AND sending sideways to AAPS
AAPS on kid's phone
AAPSClient on my phone, husband's phone (usually force closed due to battery drain), my mother's phone. Only mine is actually used to upload, the others are just reading (technically allowed to upload, but users haven't been trained to do that). 
Multiple Fitbit Versa of different generations, reading from NS but not uploading.

I will test your suggestions when kid is home from school today.

MilosKozak, 2023-08-31:
@kristinsmotz using NSCv1 or v3?

kristinsmotz, 2023-08-31:
V1 since I couldn't get v3 to work with AAPSClient

rosseland, 2023-09-01:
Nsc version is probably relevant.
I don't have this problem on 3.2rc1 using nscv3 on all phones.

MilosKozak, 2023-09-01:
resolved in https://github.com/nightscout/AndroidAPS/commit/919c79ebf9cf77b0e0ec486858092764bf8aad62

closed 2023-09-01 by MilosKozak