nightscout/AndroidAPS #2632: crush after import older settings

naamanmeir, 2023-07-20:
hello,
current system is aaps 3.0.0.1 , upgraded from 2.8 , working for about two/three years now , with rileylink/orange. all is well.

after installing version 3.1.0.3 ontop of 3.0.0.1 , the app does not load. 
installation successful but app failed to open.

after clear all data for app , cache+data, app is running and init setup is required ,from clean start , 
the app is functioning ok but all settings+progress+objectives is lost.

after importing settings exported from older version , the app again does not load , 
it writes a very short logfile and quits right away.

seems like there is a problem in the settings file that newer version rejects.

when downgrade back to clean 3.0.0.1 the import is successful , 
3.0.0.1 can import its exported settings and run again.
but 3.1.0.3 does not.

tried : 
installing beta 3.2 , exporting empty settings with only objectives , change to virtual pump , before/after importing , 
installed version 2.8 and exported settings , removing all stored preferences files and create a new with a clean profile on 3.0.0.1,
non solved the problem.

the problem : mostly the objectives actually , the settings i can manage its not a problem at all , 
only the objectives timeout is a problem.
current system is back to before attempted upgrade , 3.0.0.1 with rileylink.

thank you

MilosKozak, 2023-07-20:
you can PM me the export with master password. I can try it

naamanmeir, 2023-07-22:
i've sent you a pm on the discord channel , is that ok? any better way?

MilosKozak, 2023-08-03:
fixed by https://github.com/nightscout/AndroidAPS/commit/d9b69f59d1a3f60916de861e44bef06d2ea45412