nightscout/AndroidAPS #2638: Excessive battery usage with time / data growth

miiihi, 2023-07-27:
AndroidAPS v 3.1.0.3

After two years, my Pixel 5a managed only about 10hrs of battery time. AAPS was first in the app battery usage list and accounted for more than 20% of battery usage, but thought that's normal - that the battery is simply worn out and was already looking for new phone. 

After reading a post, that purging the AAPS data, could help with battery usage, I gave it a go. My AAPS user data storage use was reported at about 1GB. Purged it, imported back the settings and indeed, the AAPS is not first in the list anymore and phone battery now easily handles 24hrs again!

There is obviously a data access pattern somewhere that doesn't scale and should be fixed.
Alternatively, as a quick workaround, a "Purge data" functionality / button should be implemented to easily purge data older than X days... I saw the "Reset databases" button in the Maintenance, but in my quick test it didn't do anything (but could be wrong on that, was only a quick test) and also couldn't find any documentation on "Reset databases" functionality. What it does or should do...

MilosKozak, 2023-08-01:
Cleanup is already available in 3.2 beta

closed 2023-08-01 by MilosKozak

phrogg, 2023-09-10:
Interesting, I'm having this issue since, two years. I'm not coming over a day with my phone now. Today my phone was at 100% I unplugged it an 1-hour later aps already drained 14%. My battery is new. I recently changed it. And without aps my phone works for about one and half days. I tried to remove the old data like here:
> After reading a post, that purging the AAPS data, could help with battery usage, I gave it a go. My AAPS user data storage use was reported at about 1GB. Purged it, imported back the settings and indeed, the AAPS is not first in the list anymore and phone battery now easily handles 24hrs again!
But unfortunately, it didn't help. I don't know what else to do.
Does anyone have any idea?

jaredwhittle, 2023-09-13:
I'm having the same issue, where I did the Reset Databases trick, and it only marginally helped my battery. AAPS is still the #1 use app by far in the battery section. I also have NSClient set up to only upload during charging, but still, my less than 1 year Pixel 7 barely makes it to dinner time, and I'm not a huge phone user during the day either.

phrogg, 2023-09-17:
> I'm having the same issue, where I did the Reset Databases trick, and it only marginally helped my battery. AAPS is still the #1 use app by far in the battery section. I also have NSClient set up to [...]

Good to hear, that I'm not the only person affecting this. Because I have these issues since one and a half years...