nightscout/AndroidAPS #2641: 3.2.0 Beta1 Home Screen Width overzooming

CAPTCG, 2023-07-29:
Repo
[image]
[image]
rting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
- Hello. I have a problem with my Samsung S22 screen width. The AndroidAPS home screen width view zooms randomly so parts on either side are not visable. The only way to reset the screen is to "exit" the app using the three dot menu in the upper right corner. Then upon reopening the app the screen is back to the correct width. This seems to occur after the bolus completed overlay screen leaves the home screen. Any idea how to correct this. I am using the 3.2.0 beta1 version. Thanks for any help to resolve the issue. I appreciate everyone contributing. 
- Craig 
[image]

MilosKozak, 2023-08-01:
It's usually cased by setting larger font in Android. Or you can experiment with Low-res skin in AAPS. There is nothing more we can do with

closed 2023-08-01 by MilosKozak

CAPTCG, 2023-08-01:
Hello Milos,

Thank You for your reply and for your time. I did try all of the steps you
suggested.  I have the same issue even with the smallest plain font. I did
also set each of the skins one at a time to see if the issue was resolved.
The issue was just less obvious with the changes to my settings of skin and
font however the over zoom is not resolved. The zoom out of the width
immediately when the bolus screen disappears is still persistent on
occasion.  I would say this over zoom happens about once every twenty or so
times after a bolus. Not a big deal at all but I wanted to report the issue
just the same.
I know you are very involved with the AndroidAPS system development and
ongoing maintenance.  I am only involved in AndroidAPS for about 60 days
and really am grateful for you and the entire community of people working
together to better the lives of T1D people.
I am a retired construction buisness owner. Not sure how I might be able to
contribute.  If the community needs anything I may be able to spend time
working on please reach out to me.

Thanks again for everything,

Craig Gordon
732-278-2414