nightscout/AndroidAPS #2645: [EOPATCH2] Request for an alarm mute button (improvement)

fjpezuela, 2023-08-02:
It would be a good option to have a "Silence Alarms" button, like the one on the Omnipod Dash with the GlucomenDayPump (EOPATCH2).

I have had more than 3 times that I have not seen the button to confirm the alarm that the patch expiration is near or that the patch has expired. In that case, if the notification has not been confirmed, it is no longer possible to avoid the beeping of the patch.

The idea is that if for some reason, the patch is beeping, it can be silenced by the button.

@jungsomyeonggithub

Thanks!

anagomezinsausti, 2023-08-03:
It's a great idea and very useful. Thank you for thinking about it.

jbm76, 2023-08-19:
Hello!!!

It would be great if it could be implemented!

Hopefully the GlucomenDayPump (EOPATCH2) can work as if it were a Dash. 

It has also happened to me not to see the notification and continue with the beep. 

Thanks in advance for the work done.🤗🤗

> It would be a good option to have a "Silence Alarms" button, like the one on the Omnipod Dash with the GlucomenDayPump (EOPATCH2).
>
> I have had more than 3 times that I have not seen the button to [...]

scastanocaro, 2023-08-20:
Please, it is neccesary that "silence alarms" button as not for all patches the confirmation notification appears in the master phone. Thank you very much in advace.

fjpezuela, 2023-08-30:
Hello! Today it happened again that the system did not notify me at 11:25:34 that the patch was going to expire, nor at 12:25:34 that the patch has expired. 

This doesn't happen all the time, only from time to time, but when it does, there is nothing to do. The patch doesn't stop beeping until you replace it.

I'm leaving the logs here, in case you can see where the problem is.

Is it possible that this improvement will be fixed before 3.2.0 is released? There are many of us who are using this pump in Spain and we all have this problem. 

It would also be great if all the notifications that appear in AAPS could be added to the treatment page, as another option, as currently once they are received, you can't consult past notifications.

Thank you very much for all your support.

[AndroidAPS.zip](https://github.com/nightscout/AndroidAPS/files/12474776/AndroidAPS.zip)
[image]
[image]
[image]

scastanocaro, 2023-08-31:
Good morning. We have another pod with the beeping. The sound is really very disturbing anda repetitive. With this sound it is not possible to be at school or at work. I would really appretiate if you can fix it.

clagilmen, 2023-09-06:
Its a good idea. This sound its very annoying.Thank you for all your support.