nightscout/AndroidAPS #2651: Feature request: option to sync / import temp basals from Nightscout

Kenoubi, 2023-08-06:
There appear to be options for syncing basically everything else (including boluses and profile-default basals!) from Nightscout, and those work great, but I can't see any way to get AAPS to import temp basals.  This would be super useful when switching to a new phone, because you could get everything (almost) perfectly lined up on the new phone, using virtual pump with the loop disabled, and then switch over while still having accurate IoB.  (Also, the history browser on the new phone would show what actually happened, statistics on the new phone could accurately calculate total daily dose immediately, etc.)

Please let me know if there's some way to do this and I've missed it!

Thanks,
  Kenn

MilosKozak, 2023-08-07:
all data is synced with NS

closed 2023-08-16 by MilosKozak

Kenoubi, 2023-09-22:
Thank you!

Why is syncing TBRs dangerous in a way that syncing other data isn't?  I found where in the code this is and the commit that added it (https://github.com/nightscout/AndroidAPS/commit/fcd4b736d779f2cc11a88c7746305936746cfe7d) but it doesn't have an explanation as far as I can see.