nightscout/AndroidAPS #2661: URGENT : DASH communication problems. Need updated master 3.1.0.4 or 3.2 beta2

robertrub, 2023-08-11:
@MilosKozak  Several people are loosing DASH pods and have problems because of the DASH communication bug in 3.1.0.3. This bug has been solved in dev for months.

Please put out either a 3.1.0.4 with the new DASH (from memory 3 commits) code or put our 3.2beta2 so that these people can have working DASH pods.

Thanks.

jbr7rr, 2023-08-12:
Are you sure it is due to the things fixed in Dev? Do you have confirmation that these issues are resolved after updating to dev?

robertrub, 2023-08-12:
Hey, **you** did the fixing, you should know :)

Yes, of course Dev fixed the DASH connection problems and since you changed the code, all is working like a Swiss clock :)

jbr7rr, 2023-08-12:
> Hey, you did the fixing, you should know :)
>
> Yes, of course Dev fixed the DASH connection problems and since you changed the code, all is working like a Swiss clock :)

I know what it fixes. But you were referring to people losing pods. Do you have confirmation that when they (the ones who lose the pods) update to dev that that fixes their problems?

I would prefer to see some logs. I've heard some people losing pods on facebook and discord, but have not yet seen logs. Sometimes unexpected things can happen, maybe insulet changed some things in their pod FW? Or some phone update? The latter could be an indication that the fixes made could help.

Just want to make sure before giving ppl false hopes,

However if Milos agrees a 3.1.0.4 version is needed I can prepare a branch and PR for that

robertrub, 2023-08-12:
Can't be sure of course but there is a huge possibility that they are communication problems. And, you solved those.

You'll maybe see a confirmation in his logs...

MilosKozak, 2023-08-14:
no 3.1.0.4, there is so many other fixes in 3.2 ..... everyone must update

robertrub, 2023-08-14:
Than maybe a 3.2beta2 is a better option... It will permit a larger testing of the actual dev version and help all those with phones not totally ok with the old DASH connection code.

blublubbluf, 2023-08-15:
@MilosKozak is there a rough timeline of when a G7 compatible version will be publicly available? We´re having the same problems as @robertrub. Should the soulution be a 3.2beta2 how would we gain access to that?

MilosKozak, 2023-08-15:
rc1 will be soon

closed 2023-08-16 by MilosKozak