nightscout/AndroidAPS #2663: Dana I frequent lost connection

petervoogt, 2023-08-13:
The Dana I combined with the Cubot pocket has a frequent problem connecting. Today I saw the phone was finished with the connection (no activity on screen) but on the Dana I display there was still activity from the connection symbols indicating it was still busy connecting to something. I think the "disconnect message" sometimes is not recieved by the Dana I pump.

After a few minutes of no activity, the pump stops searching the other BT device and all is ok again.
I think this might also be the reason I sometimes recieve a "pump not available" message. Latest were
02:02 - 1-08-2023
04:27 - 30-07-2023
10:53 - 28-07-2023
06:10 - 27-07-2023 2x
04:45 - 27-07-2023 4x
19:24 - 22-07-2023

becouse of connection issues I factory reset/re installed oem-software not long ago, since then the connection issues are less, but not gone.

Hope you can help
Peter, Netherlands

version 3,1,0,3

[AndroidAPS_LOG_1691922459987.log.zip](https://github.com/nightscout/AndroidAPS/files/12329328/AndroidAPS_LOG_1691922459987.log.zip)

Time it happened, several times in the last minutes and during the creating of the log-files.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2023-08-16:
test 3.2 dev
there was an improvent

petervoogt, 2023-08-16:
Then I will wait, it is my daughter (10) who is using AAPS. Now that we know the problem has been identified and adressed this is not a problem.

thank you

closed 2023-08-17 by MilosKozak

petervoogt, 2025-02-17:
now after more than a year it looks like there was something wrong with the
BT hardware on the phone!!!!! Connections got better over time with updates
but still got lost frequently.

We bought an identical phone with identical software numbers and now, no
problems keeping the connection.

I guess sometimes the answers are just this simple.

Thank you and the team for all your efforts and keep up the good work. I
don't know how my life would have looked like without you.