nightscout/AndroidAPS #2676: Profile change when on pump disconnect activates basal

saFariW, 2023-08-17:
Build: dev-k (personal branch) merged at [d68869c](https://github.com/nightscout/AndroidAPS/commit/9c229e72654758f30128427fc0684d49cd68869c)
Pump: Medtrum Nano
Time: 19:15
Logs: [AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/12372103/AndroidAPS.log)

I dont believe it is intended that basal is administered after a profile change while there is a pump disconnect active.

Reproduction:
[image]

jbr7rr, 2023-08-17:
This is also reproducible with dash

jbr7rr, 2024-02-12:
@MilosKozak  Also happens on other pumps (dash)
But I will pick this one up anyway

closed 2024-02-27 by saFariW