nightscout/AndroidAPS #2685: Nsclient in different timezone causes incorrect basal

faldor20, 2023-08-21:
When I view the basal rate of my friend who is in +10 in my nsclient which is in +3 the basal displayed when there is no temp basal is the basal she has set for the hour that I'm in, not the one she's actually got. 

Eg basal at 3am is 1.2 basal at 10am is 1.5
I see 1.2 she's on 1.5

I've also noticed significant deviations in the COB between the two of us and I wonder if that's because of this as well. 

I have her timezone set in her profile  on  nightscout 

Ns: 14.2.6
Nsclient+aaps: 3.1.0.3
Logs:
[image]

faldor20, 2023-08-21:
Oh, also I tried changing the basal slightly in nightscout and I did get the updated profile so I definitely have the correct profile

faldor20, 2023-08-22:
I am in Greece with nsclinet on +3 time. She, with aaps is in Australia with +10 time. If my phone is set to Greece time, as it should be. The basal shown to me is wrong. It's based on my time in my phone. It should be based on her time on her phone, or based on the timezone set in her profile.

faldor20, 2023-08-22:
In the meantime I have manually set my phone to run Australia time. I have a watch and can make do with my phone being out.

MilosKozak, 2023-08-23:
you should see the same situation only time will be different

faldor20, 2023-08-24:
But doesn't the profile have a timezone set in it, isn't making this work the point of setting the timezone? Or is that data not available to aaps?

MilosKozak, 2023-08-24:
this is used only for rendering in NS