nightscout/AndroidAPS #2691: Omnipod Dash - option to mute alerts regarding pod needs to be changed within 8 hours

olorinmaia, 2023-08-22:
After using a pod for 72 hours the omnipod dash beeps several times during the 8 hour window to alert you to replace it. There is no option to disable this alert in AAPS today. This ticket is regading adding that option. 

I have discussed it on discord and it seem possible to add this option. Since there are alerts in Nightscout / AAPS  / AAPSclient for pod needing to be changed I find this alarm quite unecessary and disturbing as it alerts several times. 

Example today it was embarrassing for my son as it happend during class and other places when these alerts make lot of attention. I would really appriciate this being an option in AAPS 3.2.0 if possible :)

Thank you all for your hard work. AAPS is really a lifechanger for us :)

jgslade, 2023-08-23:
This definitely would be a handy setting to have.

MilosKozak, 2023-08-23:
is this possible with original remote?

jbr7rr, 2023-08-23:
> is this possible with original remote?

AFAIK No, but Eros driver does have an option to mute these alarms

> I wonder if this even possible? Are the alarms not built into the pod hardware?

Yes it is possible, Ive hardcoded this in my own branch to be disabled. I can make a proper solution with a setting for this.

@MilosKozak Given that original PDM does not support this is this wanted?
If this functionality is desired, what has the preference:
1. A single setting that just disables all pod expiration sounds
2. A setting that allows the users from when they want to have the expiration alarm? E.g. at hour 72 (default) or hour 79 etc. setting that at

olorinmaia, 2023-08-23:
If i get a vote option 2 is for me perfect! 👍😎👏

olorinmaia, 2023-09-04:
https://github.com/nightscout/AndroidAPS/commit/d6e6a986427c8ff9db464e0dee92639fc526e94e

closed 2023-09-04 by olorinmaia