nightscout/AndroidAPS #2692: Enhancement / Simplification / Expandability -  integrate with Google FIT

FlyingToto, 2023-08-22:
This overlaps with several other Issues/Enhancements... but....

- Make AAPS, nightscout, BYODA, everything leverage Google FIT as a central exchange framework.  
For instance: 
- pulling HR data from google fit will allow AAPS to consume HR data from wearOS without having to install aaps companion app, will work with other watches like skagen, fossil, garmin and any other HR compatible assets.
- access to weight/BMI/....
- access to Nutritional values  from other carb countings apps like MyNetDiary including proteins/fats 
- access to efforts/steps/energy spent during sport activities.
- access sleep patterns  
- access to Menstruation Cycles
....

Advantages:
- should drastically simplify coding integration of all the components if they all consume/access the same google fit API
- should promote open standard on android (similar to what apple does on IOS as even dexcom is publishing BG into apple health!)
- This also opens the door to perform more 2nd degree analysis over time with auto-tune... looking for patterns based on these various criteria  in the long term on a per individual basis.
- potentially allow to extract some of these individual patterns to identify more global trends (400calories of energy expanded in 2h increase Insulin sensitivity by X% for 2h on average, ??maybe REM sleeps increase ISF?? who knows??)

ArthurusDent, 2023-08-22:
Are you sure Google Fit is the correct app for the job? There is a new app called Health Connect by Google. Going by the comments in the Play Store it looks nowhere close to being usable in an environment like AAPS but in theory, it should be eventually. I've talked about it here: https://github.com/nightscout/AndroidAPS/issues/1586#issuecomment-1407396810

LuisPuertollano, 2023-08-24:
> Are you sure Google Fit is the correct app for the job? There is a new app called Health Connect by Google. Going by the comments in the Play Store it looks nowhere close to being usable in an environment like AAPS but in theory, it should be eventually. I've talked about it here: [#1586 (comment)](https://github.com/nightscout/AndroidAPS/issues/1586#issuecomment-1407396810)

Xdrip has started with integration with Health Connect,...there is an option, see:
https://xdrip.readthedocs.io/en/latest/use/interapp/#google-health-connect

vanelsberg, 2023-08-27:
Assumption is that Android devices connect to Google Health/Fit...
This may not be the case and could introduce a dependency that may limit usability?

Samsung devices (phone, watch, other) for instance uses it's own Health platform and does not integrate with Google.

ArthurusDent, 2023-08-27:
> Assumption is that Android devices and health devices integrate with Google Health/Fit...
> I think this is not the case and assuming would/could introduce a dependency that may limit usability for non-Google phones?

No idea whether Health Connect works on non-Google phones. The chances might be a bit higher than with an average app because Health Connect is explicitly built to share data _on_ the device. It appears to not share data with the cloud as the default option. Therefore it might not depend on the typical Google Services.

> Samsung devices (phone, watch, other) for instance use there own Samsung Health platform and do _not_ integrate with Google Health and would be incompatible with proposed solution.

Samsung Health is explicitly listed as a compatible app for Health Connect: https://health.google/health-connect-android/

FlyingToto, 2023-08-27:
ok so let's summarize a few things:

Google Fit 
-  just a standard app, so it should be compatible with any android phones. 
- supposedly doesn't require internet access
- tracks standard activity/Body Measurements, Vitals(including BG and HR), Nutrition, Sleep, Cycles
- no ability to record treatments etc...
- compatible with wearOS
- most apps are compatible with it  (sales, nutrition, exercise apps,...) except... 
- if using skagen/Fossil (non-wear) app, BG recording seems "spotty" from personal experience
- if using garmin (non-wear) app, 
- - not compatible with Google Fit natively. 
- -   you can get for free "activity only" data into google fit by using strava over interent only
- - you can get BG, efforst, etc... data to google fit using paid apps but by internet only

Google Health
- still a beta?? with unclear roadmap
- unsure if it can be installed on all android or only subset (on s22 looking in the google play store takes you to google fit) might be because samsung as their own samsung health... 
- samsung health compatibility is limited and seems proprietary.

conclusion (version 1): 
- might not be suitable for reading reliably live data (except for wearOS but we already have the AAPS app for that)
- could be useful for parsing large data sources for after-the-fact reporting/statistics 
- can be used to read carbs input from an array of nutritional apps (similar to iOS loop, you can then have AAPS prompt with the bolus screen and potentially leverage details inputs about proteins/fat in addition of simple carbs) 
-  can be used to track slow changing data (weight, bmi, Weely/daily exercises... ).
- in the spirit of open-source, if we were to integrate with it, we could offer to upload stats into it like BG from AAPS

vanelsberg, 2023-08-27:
> Samsung Health is explicitly listed as a compatible app for Health Connect: https://health.google/health-connect-android/

As far as I know this only enables Google to access data in Samsung Health. So this would not help.

Also proposed solution raises another important issue (at least for me):
This would force users to to share AAPS and health data with Google. As I absolutely do not trust my data with Google (only allowing the minimum needed to operating the OS) this would not be an acceptable option:

I would never allow sharing this kind of private data with Google or any other cloud service, not my health data from Samsung health and especially not my data from AAPS, BYODA, Nightscout and others.

I would strongly suggest to make any integrations like this an option, and let users decide how much they value their privacy.

vanelsberg, 2023-08-27:
> Make AAPS, nightscout, BYODA, everything leverage Google FIT as a central exchange framework

Wondering if this would work?
I expect delays in syncing realtime data like HR, stepcounts and other relevant distributed data with the platform which could make realtime interaction with AAPS at least problematic?

That said. I can see a service as Google Fit as een central place where data is stored, replacing Nightscout/Tidepool e.a. solutions.

FlyingToto, 2023-08-27:
> 

yes that's what I mentioned in the summary...

FlyingToto, 2023-08-27:
> I would never allow sharing this kind of private data with Google 
that boat sailed a long time ago as soon as you use android... if you want privacy you would need to get back to BSD's hardened built  running podman (possibly some linux) isn't it?

ArthurusDent, 2023-08-27:
> * just a standard app, so it should be compatible with any android phones.

I believe @vanelsberg talked about de-Googled phones, not about Android compatibility. De-Googled phones are Android phones that don't use any software from Google to stop Google from spying on the user. 

And even if we were concerned about Android compatibility, Health Connect only requires Android 9.

This FAQ answers important questions and everybody should read it: https://developer.android.com/guide/health-and-fitness/health-connect/frequently-asked-questions 

For instance it says that even Health Connect needs Google Play Services. But there is something like MicroG, that implements Google Play Services in a non-proprietary way. Maybe this would enable Health Connect on de-Googled phones. I don't have a de-Googled phone so I'm not an expert on this.

There is a table that compares Health Connect and Google Fit: https://developer.android.com/guide/health-and-fitness/health-connect/frequently-asked-questions#health-connect-vs-fit

@vanelsberg said: "As far as I know this only enables Google to access data in Samsung Health. So this would not help."

As far as I understand Health Connect, it's up to the user who gets the data. The app collects and shares health data with apps _on_ the device. The user gets to decide which apps get the data, which means that Google doesn't get the data, if the user doesn't want to share it with Google. Google doesn't get the data just by Health Connect having it. Therefore your statement is incorrect. 

This means that Google will also not get your AndroidAPS, BYODA, Nightscout etc. data. Everything stays on the device with Health Connect, unless the user shares data with apps that upload to the cloud. This is at least how I understand Health Connect.

The biggest problems I see with Health Connect are the beta status and the slow availability of the data. No idea whether this was improved in recent versions.

vanelsberg, 2023-08-27:
> > I would never allow sharing this kind of private data with Google
> that boat sailed a long time ago as soon as you use android...

Yes :-) But no BSD for me. Imho, that would even more unsafe...

I run a Samsung phone which both connects to Google and Samsung cloud and its own Samsung Health platform.
That does not mean I just "share" because I can or some app needs access.

Don't get me wrong. For me keeping personal data and privacy is important. So sharing only when absolutely necessary - or take a calculated risk (based on the type of data) when I see benefit in sharing.

My intended argument:
I do not like the idea of a dependency of AAPS and accompanying apps on a 3th party service (google or any other) that needs access to the users private data. That is - _it should always be an option where the user can decline to use without limiting AAPS's functionality_,

vanelsberg, 2023-08-27:
> As far as I understand Health Connect, it's up to the user who gets the data.

I see this seems to be true for "Health Connect". Was not aware of this. But then: what is the point in using it when all is only on the phone?

**Google Fit** is described as:
_Google Fit is account-centric. _Data is stored on a server-side infrastructure_. Its data model is entirely tied to specific Google accounts as opposed to the user's devices. This makes the server the source of truth in Google Fit._

From a technical standpoint SOT is just what we need. Problem is it involves sharing data with a 3th party like Google.

vanelsberg, 2023-08-27:
@FlyingToto @ArthurusDent 
For clarity: From a technical standpoint I do like the idea!
I just think privacy concerns and user choice should be part of the solution?

FlyingToto, 2023-08-28:
I guess we could always take a minimal opt-in approach... 

- out of the box, AAPS can try to "read" data available from google fit/health... (no privacy issues there)
- people can opt-in to "write" data into google fit/health (and pick and choose which one to share, live or not) 

TBH, I think the primary use case would be just for the nutritional inputs, (not sure how aaps can improve its isf/ecarb calculations if it gets access to more detailed proteins & fats in addition to carbs)...

ArthurusDent, 2023-08-28:
> I see this seems to be true for "Health Connect". Was not aware of this. But then: what is the point in using it when all is only on the phone?

Health Connect converts the data that different incompatible health apps have into a common standard and then users can share the converted data with other apps which might be able to provide additional benefits when using that converted data. My initial thought was to use this for period tracking: https://github.com/nightscout/AndroidAPS/issues/1586#issuecomment-1407396810 Women can just use their preferred period tracker and tell Health Connect to share that data with AAPS. All on device, no sharing with Google.

I still don't really get why Health Connect is non-cloud. Is sharing data with different other cloud apps easier because the clouds convert the data? Or is that also a missing feature for cloud apps? I don't know how Google Fit works in that aspect. Maybe enough people want to share their health data with other apps but Google knows that people don't want Google to then have all that data. So Google came up with a non-Google-cloud solution, which is Health Connect.

There of course is that problem with smartphone loss or switching to a new one. People might even want cloud upload but probably to someone that isn't Google. We do already have Nightscout. No idea what's planned on their side. 

Also, you might want to have as much data as possible with upcoming AI enhanced dosing algorithms, which is why you probably really want to trust your data is safe somewhere in the cloud and the likelihood of data loss being minimal.

Btw: Not only can Google Fit talk to Health Connect, but also my installed xdrip nightly from mid of July. You can at least get a first impression how data sharing on device would work without risking uploading data to Google.

schmadde, 2023-09-02:
This is a topic that I am very interested in, since exercise is the one area where the loop fails completely and manual treatments are difficult. Retroactive analysis quite hard because data is not combined in one place. 

My thoughts on this are that google fit as only solution is not good. I do not want to post my health data in yet another cloud service - and no, "your data souvereignity is lost anyway" is not a good way of thinking. I prefer the "need to know" principle which has proven quite well.

I personally think this data collection belongs into one of the data hubs like xdrip+ and/or nightscout. I would prefer xdrip+, since this works without network connection (and during exercise I am quite often in areas with bad or no network coverage or outrageously expensive one), but nightscout is good for analysis.

For AAPS I think a generic Interface would be good which would then poll data from NS or get it pushed via inter app communication from xdrip+. 

I would also be interested in bulk data loading for the case when you actually use an exercise file for the last 3 hours or so and let it be consumed by the loop.

The question then is: what should the loop then do with it? Can any of the algorithms make decisions based on HR? 

My experience with exercise is that the loop works against me during the exercise: it pumps insulin when I start  high which is counterproductive, because the BG would fall anyway and the added insulin would then lead to a hypo eventually. After 1-2 hours it would stop insulin completely which can lead to a keto eventually. Setting the target high and the profile low does not really help as I saw yesterday. Just setting the pump to 30% and the loop off works much better. 

So I think a good start would be something that can handle past exercises and only care for the increased sensitivity (and also clear negative IOB). For unplanned exercise like working in the garden or moving stuff around "live data" would be cool, however.