nightscout/AndroidAPS #2706: MINOR : RC1 HR lower limit is NOT 40

robertrub, 2023-08-25:
@MilosKozak Your commit sets HR graph lower limit to 40 but it does not seem to work.

[image]

[image]

Version is last Dev + PR 2700 for testing DASH alert suppression 

[image]

mtanjagithub, 2023-08-25:
Sorry why 40bpm? I have a resting heart rate of 34 bpm.

robertrub, 2023-08-25:
_I am an avid cyclist_

That's probably the explanation ;) 

The min can be put to 30 it it seems ok to you (or even better, automatically detected, or user defined...)

mtanjagithub, 2023-08-25:
30 that will work!

MilosKozak, 2023-08-25:
> @MilosKozak Your commit sets HR graph lower limit to 40 but it does not seem to work.
> 
> [image]
> 
> [image]
> 
> Version is last Dev + PR 2700 for testing DASH alert suppression
> 
> [image]

you are not on latest dev

MilosKozak, 2023-08-25:
https://github.com/nightscout/AndroidAPS/commit/98fa5ef911b5d326330854abf0bbd3f27fbaa468

closed 2023-08-25 by MilosKozak