nightscout/AndroidAPS #2713: [3.2.0-RC1] Medtrum Nano, Double history entry

jbr7rr, 2023-08-28:
Note: Got this report from another user via PM

Build: 3.2.0-RC1
Pump: Medtrum nano
NSClient V1
Time:  13:36 and 14:46
Logs:  [AndroidAPS._2023-08-28_02-00-09.zip](https://github.com/nightscout/AndroidAPS/files/12454706/AndroidAPS._2023-08-28_02-00-09.zip)

Bolus was doubled in treatments, I checked the logs for errors in the pump sync, as far as I can see there werent any faults with the driver and the bolus was reported as expected:

First the bolus is temporary synced, After bolus is finished, the record is updated. (This is the usual happy flow for the Medtrum driver)

```
13:46:31.740 [Thread-28141] D/DATABASE: [PumpSyncImplementation.addBolusWithTempId():189]: Inserted Bolus Bolus(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=MEDTRUM, pumpSerial=d1b2d5e, temporaryId=1693223191722, pumpId=null, startId=null, endId=null), timestamp=1693223191722, utcOffset=7200000, amount=0.2, type=SMB, notes=null, isBasalInsulin=false, insulinConfiguration=null)
13:46:31.744 [Thread-28141] D/PUMPCOMM: [MedtrumService.setBolus():349]: set bolus: **NEW** EVENT BOLUS (tempId) 8/28/23 13:46 (1693223191722) Bolus: 0.2U 
13:46:31.811 [RxCachedThreadScheduler-59] D/WEAR: [DataLayerListenerServiceMobile.sendMessage():200]: sendMessage: /rx_bridge {"type":"info.nightscout.rx.weardata.EventData.Status","externalStatus":"","iobSum":"1.58","iobDetail":"(2.62|-1.04)","cob":"0g","currentBasal":"2.40U/h","battery":"51","rigBattery":"51%","openApsStatus":1693223189350,"bgi":"-0.2","batteryLevel":1}
13:46:32.255 [binder:10741_C] D/PUMPBTCOMM: [BLEComm$mGattCallback$1.onCharacteristicChanged():248]: onCharacteristicChanged data: [32, 34, 16, 0, 1, 0, 94, 15, 0, 0, 0, 0, 0] UUID: 669a9120-0008-968f-e311-6050405558b3
...
13:46:42.542 [binder:10741_C] D/DATABASE: [PumpSyncImplementation.syncBolusWithTempId():211]: Updated Bolus Bolus(id=68921, version=1, dateCreated=1693223202532, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=MEDTRUM, pumpSerial=d1b2d5e, temporaryId=1693223191722, pumpId=1693223190000, startId=null, endId=null), timestamp=1693223190000, utcOffset=7200000, amount=0.2, type=SMB, notes=null, isBasalInsulin=false, insulinConfiguration=null)
13:46:42.545 [binder:10741_C] D/PUMPCOMM: [GetRecordPacket.handleResponse():143]: from record: EVENT BOLUS 8/28/23 13:46 (1693223190000) Bolus: 0.2U 
```

As the user has enabled receive Insulin from NS, I suspect this might have something to do with this? @MilosKozak Can you check this on NS / Database side? 
Also note I have seen a doubled bolus once with this option enabled on NSClient V3, but at that time I had only pump logs enabled, I haven't had this myself after that one instance.

[image]
[image]
[image]

jbr7rr, 2023-08-28:
Thinking about it there is one thing I can do from the driver:

https://github.com/nightscout/AndroidAPS/blob/b9975d4105d842e1546aa52a591e9eda495488d9/pump/medtrum/src/main/java/info/nightscout/pump/medtrum/comm/packets/GetRecordPacket.kt#L118C47-L118C47

```
                                val syncOk = pumpSync.syncBolusWithTempId(
                                    timestamp = bolusStartTime,
                                    amount = bolusNormalDelivered,
                                    temporaryId = detailedBolusInfo.timestamp,
                                    type = detailedBolusInfo.bolusType,
                                    pumpId = bolusStartTime,
                                    pumpType = medtrumPump.pumpType(),
                                    pumpSerial = medtrumPump.pumpSN.toString(radix = 16)
                                )
```

Change  ``timestamp = bolusStartTime,`` to ``timestamp = detailedBolusInfo.timestamp`` 

@MilosKozak would this be a desired and proper fix for this problem?

MilosKozak, 2023-08-28:
this is serious. conflict of  syncWithTempId and NS back sync 
having receiving from NS disabled by default has a good reason

i have to think about it

Note for me: NS back sync is using timestamp

MilosKozak, 2023-08-29:
the issue is valid for all tempId plugins
It has to be resolved in NS plugin

jbr7rr, 2023-08-29:
Cought the error today on my own setup with NSClient V3 and the Setting "Receive Insulin" enabled
Treatments in NS were also doubled

@14:53

[AndroidAPS._2023-08-29_08-18-05_.30.zip](https://github.com/nightscout/AndroidAPS/files/12466037/AndroidAPS._2023-08-29_08-18-05_.30.zip)

[image]
[image]

MilosKozak, 2023-08-29:
hopefully resolved by https://github.com/nightscout/AndroidAPS/commit/394883165d2f6603286d7bb56f6312d8b742c3b8
open new issue with new logs if not

jbr7rr, 2023-08-30:
Nice, SMB now only gets uploaded when bolus is fully synced, works as expected so far. Thx