nightscout/AndroidAPS #2715: Broken connectivity to Dana R

viq, 2023-08-28:
Somewhere between https://github.com/nightscout/AndroidAPS/commit/1d6a48d01dcd16a5767940f9e04e3003af60e713 and now (build `0105d24869-2023.08.28`) connectivity to my Dana R was broken. Currently even clicking on the bluetooth icon in the pump status does not result in reconnection. Attaching logs.
[AndroidAPS._2023-08-28_22-06-43_.342.zip](https://github.com/nightscout/AndroidAPS/files/12458376/AndroidAPS._2023-08-28_22-06-43_.342.zip)
[AndroidAPS._2023-08-28_22-06-43_.343.zip](https://github.com/nightscout/AndroidAPS/files/12458377/AndroidAPS._2023-08-28_22-06-43_.343.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/12458378/AndroidAPS.log)

Korostensky, 2023-08-29:
I have the same problem. Dana Rv2 pump.

MilosKozak, 2023-08-29:
fixed by https://github.com/nightscout/AndroidAPS/commit/f16229c4ed3f8e33c297f6c03cd6200e31df933b

closed 2023-08-29 by MilosKozak