nightscout/AndroidAPS #2726: BUG: Double down arrows are back

robertrub, 2023-09-01:
Hi @MilosKozak 

Sorry to reopen this subject that was fixed 😕 but it happened again with the last dev-l +PR2700

I don't know if it is important but ATM, I have 2 G6/Anubis BG sensors running with xDrip and xDrip variant1. For xDrip variant1, all propagation and updates and dlertsare turned off.

[image]

[image]

The logs

[image]

robertrub, 2023-09-01:
Once stuck, the arrows stay double down

[image]

robertrub, 2023-09-01:
Change compared to _before you had corrected_ the bug the first time. This time, they stopped by themselves without any action from me.

[image]

robertrub, 2023-09-13:
Again... The problem is that I can't see what triggers it.

[image]

einl, 2023-09-20:
I'm having the same issue (just now, first time I see this).
[image]
[image]

einl, 2023-09-20:
[image]
[Uploading AndroidAPS_LOG_1695216752874.log.zip…]()

robertrub, 2023-09-20:
Are you using any _smoothing_ in AAPS ? What is your CGM and app ? xDrip ?

einl, 2023-09-20:
Average smoothing in AAPS
Dexcom G6 on xDrip.

robertrub, 2023-09-20:
Exactly my settings too.

robertrub, 2023-10-02:
Stopped by itself (same as last time)

[image]

robertrub, 2023-10-03:
Once more. Started and ended all by itself...

[image]

parapenT1sta, 2023-10-03:
I still have G5 sensor and I had the same issue but I could find out that sensor wasn't OK. And was only in one or two readings

CaroGo, 2023-10-22:
I can confirm this issue...using G6 average smoothing,or no smoothing..latest AAPS Dev3.2...all deltas positive, but double arrow down is showing

parapenT1sta, 2023-10-25:
@MilosKozak , I had it today for a few readings.
Logs: 
AndroidAPS: [AndroidAPS_LOG_1698252885835.log.zip](https://github.com/nightscout/AndroidAPS/files/13168741/AndroidAPS_LOG_1698252885835.log.zip)
xDrip: [exportCSV20231025-165704.zip](https://github.com/nightscout/AndroidAPS/files/13168745/exportCSV20231025-165704.zip)

[image]

[image]

[image]

CaroGo, 2024-02-23:
[image]
Still these irritating arrows...
Version 3.2.0.3

robertrub, 2024-04-13:
I wonder if the double arrows are related to the age of the sensor... It often happens towards the end of 10 days.

[image]
[image]
[image]

MilosKozak, 2024-12-04:
when no arrow is provided, value is calculated