nightscout/AndroidAPS #2735: Minor : About/ Show NS version

robertrub, 2023-09-04:
Actually, in the _About_ window, NS version is only shown if AAPS is connected to NS at the same time. 

If NS connection is set to "connect when charging" in AAPS, even if the phone was charging 5 minutes ago (but is no more), the About screen forgets the NS version.

As NS version changes very rarely, it would be nice if the About window shoved the last NS version AAPS had detected.

[image]

[image]

vanelsberg, 2023-10-12:
Confirming. Seeing the same on dev-m
Repro: switch off: connect settings -> On battery (while not charging)

robertrub, 2023-10-20:
@MilosKozak Has there been work on this issue ? The problem has been solved.

I don't know if installing NS v15 _MASTER_ fixed it (was using NS v15 _dev_ till now) or if you fixed the code.

No upload to NS at the moment and NS version is still visible in _About_👍

Do I close this issue?

[image]

vanelsberg, 2023-10-20:
> @MilosKozak Has there been work on this issue ? The problem has been solved.

Confirming!

robertrub, 2023-11-11:
Hi @MilosKozak 
I don't know why you had reopened this issue. It has been solved. Now, once AAPS has connected to NS, the NS version info stays put in about and is visible even when not uploading.
I'm closing it.

closed 2023-11-11 by robertrub