nightscout/AndroidAPS #2748: [DEV] AAPSclient occasionally misses downloads

szantos, 2023-09-07:
The data is there in NS, but not everything gets downloaded. Sometimes SGVs (entries) are missing. I'm not sure, if it's an AAPS or NS issue.

The particular case in the screenshots/log has happened this morning after turning on the AAPSclient device, but in rare cases it can happen during the day, too when the device is on. Missing data won't get backfilled, even if I force close AAPSclient, delete cache and restart NSClient.

AAPS 3.2.0-dev-l 5c34184723-2023.08.30 with NSClientV3+websockets
NS dev v15: [d602de1](https://github.com/nightscout/cgm-remote-monitor/commit/d602de1c474c212ab8f27d17d18ce12d5679c990)
LOG: [Log Export.txt](https://github.com/nightscout/AndroidAPS/files/12545451/Log.Export.txt)

[image]

[image]

MilosKozak, 2023-09-07:
what is the timestamp of missing bg?

MilosKozak, 2023-09-10:
do they backfill when you reset sync on aapsclient? provide logs from backfill if not

MilosKozak, 2023-09-10:
and reset logging to default before

szantos, 2023-09-11:
> do they backfill when you reset sync on aapsclient? provide logs from backfill if not

How do I perform "reset sync"?

szantos, 2023-09-11:
This morning it happened again:
- on the left BG is missing
- that 25g was a bolus wizard entry: CH and bolus wizard details got synced, but no insulin

[image]

[Log Export 2023-09-11_1.txt](https://github.com/nightscout/AndroidAPS/files/12572819/Log.Export.2023-09-11_1.txt)
[Log Export 2023-09-11_2.txt](https://github.com/nightscout/AndroidAPS/files/12572817/Log.Export.2023-09-11_2.txt)

MilosKozak, 2023-09-13:
resolved in https://github.com/nightscout/AndroidAPS/commit/d379b769d3739b75bb02d07d300f41871cb18c6d

closed 2023-09-13 by MilosKozak