nightscout/AndroidAPS #2753: UI/UX Wear Management of Settings

Philoul, 2023-09-07:
Now in latest release all the Watch settings are grouped at the same place within Watch
- You acces to all main settings from AAPS menu (on Watch)
- And some additional settings, dedicated to each Watchface, can be reached with a long press on Watchface from 2 to about 8 according to selected Watchface)

But the overall number of settings are important (currently 27 in main menu) and maybe not easy to access for everybody
See #2752 with associated screenshot

This issue is for discussion for future Improvement on an UI/UX point of view...

jbr7rr, 2023-09-07:
https://github.com/nightscout/AndroidAPS/pull/2752 is a good step forward, good job.

Ultimately my personal preference is to be able to control the settings on the phone or both.

I'm not sure if there is a way that we can duplicate a settings menu, so when one adds a wear setting it only has to be done  in one place.

Philoul, 2023-09-08:
Difficulty when you have 2 settings at different places is the synchronization...
Watch and phone are not 100% connected together, so you cannot be sure when you modify on one side, that it will be modified on the other side...

First step was to be independant to data (now whatever where the setting is managed, the set of data sent to the watch is the same and has everything, so you see immediatly the result after parameter modification)...