nightscout/AndroidAPS #2755: BUG: Double bolus given

scimone, 2023-09-08:
Hello,

I'm reaching out on behalf of a friend who is currently facing the following issue:

On two separate occasions, she observed that after administering a bolus through the bolus calculator, an SMB was delivered just 5 minutes later, with the exact same dosage as the bolus. 
The SMB amount is very high compared to the basal rate. So essentially the bolus was delivered 2 times within 5 minutes, and this seems like a dangerous bug to me. 
Furthermore, the SMB settings can't be accessed, because when clicking on "Preferences", the screen turns black and the app crashes.
Is anyone able to help us out?

For reference, my friend is using the Glucomen Day EOPatch2 pump, running Android APS version 3.1.0.3-dev-f, build number 70b61f3f5a-2023.02.05, on a Samsung A33 device with Android 13.

I'm attaching screenshots of the treatment history from both occasions, as well as the extracted logs.

Any assistance or guidance is greatly appreciated!

[image]
[image]

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/12563591/AndroidAPS.log)

parapenT1sta, 2023-09-08:
Why using an AAPS Old version and not the newest version?
I think all bug reports should be related to the latest DEV version.

Ambrale, 2023-09-08:
Because in the newest version there is not our pump: EOPATCH2.

mtanjagithub, 2023-09-09:
This version is the newest, and whit EOpatch 2
3.2.0 DEV-l

MilosKozak, 2023-09-09:
this is unsupported version. update to latest. it's probably fixed

closed 2023-09-09 by MilosKozak