nightscout/AndroidAPS #2770: AAPSclient could read IOB, COB, activity from devicestatus.json instead of calculating

szantos, 2023-09-12:
AAPS (master) calculates IOB, COB, activity and uploads it to devicestatus.json in NS. AAPSclient could read and display this data, instead of doing the calculation on its own.

If not every treatment gets synced, AAPSclient displays incorrect values, which can lead wrong decisions. See "missing insulin" here - follower could think, there was a missed bolus:
https://github.com/nightscout/AndroidAPS/issues/2748#issuecomment-1713246889

The benefits would be:
- higher consistency
- less mistakes
- less workload for the AAPSclient device
- (ideally AAPSclient should not calculate anything, to ensure 100% consistency with master)

[image]

MilosKozak, 2023-09-12:
This is not how aapsclient works. It internally replays data on virtual pump.

Every solution has it's pros and cons. Depending on device status would absolutely no data without connection or not up to date device status.

Solution is to find the reason for missing data. Not changing this behavior.

szantos, 2023-09-13:
@MilosKozak 
As long as the client calculates something on its own, it can happen, that it produces another result than the master - implications are obvious. (If the cause is a missed treatment, which got not synced, or something else is indifferent - the missed treatment was only an example.)

Bigger problem is, that the client does not even know, that it displays an inconsistent situation.

Proposal:
- taking the results from master
- if these are not available, or too old, fallback to current method and at least display a warn signal
(similar to SGV: if too old, it gets crossed out)