nightscout/AndroidAPS #2774: Hidden sub-menu in settings (can only be found via search function)

blochj, 2023-09-13:
Hi all,
Not sure if this is the right place, but I found a sub-menu that only appears if I search for it (funnily enough only without the last 3 letters... See different screenshots) but not when I just scroll to the respective section. Sorry but my app is in German and unsure if same issue exists in other languages or maybe even is meant to be a feature rather than a bug 🙂
Now on RC1 but same behaviour already occured on 3.1.0.3.

[image]

[image]

[image]

[image]

MilosKozak, 2023-09-13:
this is bug in android Preferences library. Some preferences are hidden but it's possible to get there by search, clicking forward and back etc
so far there is nothing we can do except write own library

closed 2023-09-13 by MilosKozak

blochj, 2023-09-13:
All clear - thank you for the swift feedback.
If the bug affects the same sub-menus for everyone, would it make sense to list them in the wiki/documentation? I found it more by coincidence... Thank you