nightscout/AndroidAPS #2775: Nightly crashes (3.2.0.dev-l build 1caa199e5a)

elgrupo123, 2023-09-13:
We regularly have an AAPS crash at night (once a week). Today even twice. 3.56AM and 10.29AM. After that, the loop no longer works (only Basal continues to run) until you use the cell phone normally again.
Pixel 4a5G.
Thank you.

[AndroidAPS._2023-09-13_10-39-32_.9.zip](https://github.com/nightscout/AndroidAPS/files/12598431/AndroidAPS._2023-09-13_10-39-32_.9.zip)

MilosKozak, 2023-09-13:
log doesn't contain mentioned time

elgrupo123, 2023-09-13:
sorry, here are the correct logs
[AndroidAPS._2023-09-13_10-39-32_.5.zip](https://github.com/nightscout/AndroidAPS/files/12598886/AndroidAPS._2023-09-13_10-39-32_.5.zip)
[AndroidAPS._2023-09-13_03-06-20_.1.zip](https://github.com/nightscout/AndroidAPS/files/12598887/AndroidAPS._2023-09-13_03-06-20_.1.zip)

MilosKozak, 2023-09-14:
`03:56:03.471 [main] W/PUMP: [MedtrumPump.<init>():289]: Error decoding basal profile from SP: 0`
@jbr7rr this is crashing the app

MilosKozak, 2023-09-14:
there is no reason visible in logs
app is probably killed by android. for example out of memory error.
there is nothing I can help with

closed 2023-09-14 by MilosKozak