nightscout/AndroidAPS #2776: [DEV] Manual meal bolus not uploaded to NS

jbr7rr, 2023-09-13:
Build: Personal build at f271c5a92d42bceea01659d57422c36e45b4898e
Pump: Medtrum Nano
Time:  18:34
Logs:  [AndroidAPS_2023-09-13.zip](https://github.com/nightscout/AndroidAPS/files/12600286/AndroidAPS_2023-09-13.zip)

During patch change I did a manual bolus, after patch activation I entered the treatment into AAPS.
It was not uplaoded to NS.

Might be an unwatned side effect of 394883165d2f6603286d7bb56f6312d8b742c3b8 ?
[image]
[image]

MilosKozak, 2023-09-14:
it is
please test with https://github.com/nightscout/AndroidAPS/commit/d302e6eb55f648144339260bd678b4e3195eb3bd

jbr7rr, 2023-09-14:
Thx works as expected now

closed 2023-09-14 by jbr7rr