nightscout/AndroidAPS #2781: [DEV] Profile switch takes long time to sync in AAPSClient v3+WS

olorinmaia, 2023-09-14:
Step by step
1) Enact temp profile 120% duration 30 min in AAPSClient. time: 23:31
2) Sync to NS/AAPS fast.
3) Takes about 5 minutes for enacted profile switch to show in AAPSClient
4) Switch back to 100% profile in AAPSClient. Time: 23:37
5) Sync to NS/AAPS fast.
6) Takes about 5 minutes for enacted profile switch to show in AAPSclient

Expected behaviour is for profile switch to show in AAPSClient much faster, similar to V1 API.

Logs AAPSClient:
[AndroidAPS_LOG_1694727749738.log.zip](https://github.com/nightscout/AndroidAPS/files/12613551/AndroidAPS_LOG_1694727749738.log.zip)

Logs AAPS:
[AndroidAPS_LOG_1694727788503.log.zip](https://github.com/nightscout/AndroidAPS/files/12613549/AndroidAPS_LOG_1694727788503.log.zip)

[image]

MilosKozak, 2023-09-16:
this is common behavior. enacting profile may take some time

olorinmaia, 2023-09-16:
Profile change done in AAPSClient syncs to NS/AAPS ASAP, which is good. It's just that the speed of showing in AAPSClient that the profile actually changed in AAPS  is slower in v3 API then v1 API. I'm not sure if it's a big problem tho, just thinking if possible should be improved to be on par with v1. :)

olorinmaia, 2023-09-16:
I'm getting alot of spam in NSV3 log when enact profile change some times. Maybe that's whats causing this. Attaching screenshot and log. Been going like this for 10 minutes.

Tested with:
AAPSClient 3.2.0-dev-l
Build: f717d7a0b2-2023.09.16

[AndroidAPS_LOG_1694900425744.log.zip](https://github.com/nightscout/AndroidAPS/files/12641610/AndroidAPS_LOG_1694900425744.log.zip)
[image]
[image]

olorinmaia, 2023-09-16:
I managed to stop the spam by changing to V1 API on AAPS, then V1 API on AAPSClient. Then changing back to v3 API spam so far hasnt started again

olorinmaia, 2023-09-16:
I'm using Random BG and upload to NS and backfill with sync, only way to get proper readings in AAPS it seems. May not be ideal for testing?

olorinmaia, 2023-09-17:
In dev v1/v3 doesnt matter so much it seem. Try compare with 3.1.0.3 NSClient profile change vs DEV AAPSclient profile change.

Seems like the profile gets visually changed faster in 3.1.0.3 AAPS then after that syncing fast with NSClient.

In dev it says "profile set OK" asap after profile change in AAPSClient, but it can take abit of time before AAPS internally visually change profile and then sync to AAPSClient.

MilosKozak, 2023-09-20:
https://github.com/nightscout/AndroidAPS/commit/7cb27fced0a6260bc9486f005b7ab779dade38c8