nightscout/AndroidAPS #2782: [DEV] "PH" gets added in Treatments history when manual bolus (not delived by pump)

olorinmaia, 2023-09-14:
AAPSClient 3.2.0-dev-l
Build: https://github.com/nightscout/AndroidAPS/commit/471ed28a04ccd93afb6d2f33d1c7fad3c11ef640-2023.09.14
Flavor: aapsclientrelease
Nightscout version: NS15 dev

Step by step
1) Bolus from AAPSClient or AAPS, toggle "do not bolus, record only".
2) Successful sync, but "PH" gets added to treatment history

Expected result would be only "NS".

[AndroidAPS_LOG_1694728318735.log.zip](https://github.com/nightscout/AndroidAPS/files/12613848/AndroidAPS_LOG_1694728318735.log.zip)

[image]

[image]

jbr7rr, 2023-09-15:
Yes I see the same with hardware pump

@MilosKozak 

Is it an idea to make the original fix https://github.com/nightscout/AndroidAPS/commit/394883165d2f6603286d7bb56f6312d8b742c3b8 include a check if there is a temporaryId and pumpId is null to then exclude it ?

MilosKozak, 2023-09-16:
resolved in https://github.com/nightscout/AndroidAPS/commit/275e52a546791ff044121bce6984a778b724a2c6

olorinmaia, 2023-09-18:
@MilosKozak Could you apply the same fix to the Calculator in AAPSClient. If I do a bolus via the Calculator in AAPSClient "PH" gets added in treatment history as well. Maybe check Wear AAPSClient calculator also.

MilosKozak, 2023-09-22:
https://github.com/nightscout/AndroidAPS/commit/ccabee0399d33427ffa8837e53d6756d569e99b5

closed 2023-09-22 by MilosKozak