nightscout/AndroidAPS #2790: [DEV+master] Wear tiles, menus and settings aren't translated, still when there are translated strings for it?

olorinmaia, 2023-09-16:
In wear app for AAPS / AAPSClient (master and also dev) there are missing translations for almost all icons in the tiles, text in menu and settings although I think everything is translated in e.g. Norwegian. 

I notice that for example "Cancel" is translated to "Avbryt" in Norwegian but nothing else. in the temp-target tile. All other tiles / settings are in English. If I activate an item in any tile, the following screen up to confirmation isnt translated, after confirmation some translation is done.

My main AAPS has Norwegian language, and clock and phone is Norwegian language (if it matters)
[image]
[image]
[image]
[image]

olorinmaia, 2023-09-20:
@MilosKozak 

Updated to  AAPS 3.2.0-dev-l c52c0e1125-2023.09.20 and updated also wear APK and did a fresh install on watch.

I can't see any change in issue described above originally. If I change language in AAPS to English "Avbryt" turns to "Cancel" in the tiles, but all tiles except of that one stay in english no matter what language in AAPS. 

[image]

MilosKozak, 2023-09-22:
I agree it doesn't work but it works for other languages tested. Probably android bug

olorinmaia, 2023-09-22:
Cant see any difference if I change language in AAPS go Swedish for example. Or do I need to change system language on phone?

I thought language setting in AAPS decided what getting translated in app and wear. If that is true this also happen for Swedish, German I tested just now.

olorinmaia, 2023-09-22:
I notice if I change language in android to Swedish and use system default language in AAPS all translations in AAPS and wear turns Swedish. So that means system language in android does matter.

If I use Norwegian as system language in android only small parts gets translated in AAPS and wear if I use language = system default in AAPS. 
Is there some reason AAPS don't support Norwegian as system language vs Swedish or is it just a android bug?

olorinmaia, 2023-09-22:
[image]
Norwegian got three choices in android. It's Norwegian bokmål that is 100% right in most cases. In comparison Sweden only got one option.

MilosKozak, 2023-09-23:
translation is done for language marked 2 on your screenshot. Internation code "no"

MilosKozak, 2023-09-23:
[image]

olorinmaia, 2023-09-23:
This is wrong since 99,9% use number 1 in my list and also standard setting when setting up a phone in Norwegian.

Norwegian bokmål should be chosen. If possible all those Norwegian choices should recieve til translation in Norwegian.

MilosKozak, 2023-09-25:
well I can enable Norwegian Bokmal in Crowdin but it will mean new translation

olorinmaia, 2023-09-25:
They can't be merged or copyied over? So we basically have translated for a language that doesnt get recognized via the default system language in android for Norway. Workaround is now to select "Norwegian" as language in AAPS. but that doesnt translate all correctly in wear as this issue describes.

Is it possible to edit the current translation to Norwegian bokmål instead and see if that fixes this issue so we dont have to translate everything again? And also not continue translating for a Norwegian language noone uses? ;)

olorinmaia, 2023-09-27:
I noticed something in the code, not sure if it matters. 

Where translation doesn't work it looks like this: 
[image]

Where it works it looks like this: (nb)
[image]

Philoul, 2023-10-01:
@MilosKozak I asked @olorinmaia to create within wear resources a new subfolder `values-nb-rNO` and copy paste in it the string file included into `values-no-rNO`
=> It fixed the tiles translation issue for him...
So I think enabling Norwegian Bokmal in Crowdin will fix this issue...

olorinmaia, 2023-10-01:
Thanks @Philoul  :) I solved the issue locally, so I'm happy. But for other Norwegian users it would be nice to find a permanent solution. 

We know now that "values-nb-rNO" translates correctly (Norwegian Bokmål). Is there any way to copy all translated strings already done in Norwegian over to Norwegian bokmål? 
If not, I suggest we add it, and I start translating all wear strings, and leave the rest for when I got time. (Since you can force norwegian within AAPS by selecting Norwegian language)

olorinmaia, 2023-10-03:
Norwegian bokmål added to crowdin and uploaded the "old" Norwegian translations. Going to build and check that this fixes this issue tomorrow.

[image]

olorinmaia, 2023-10-04:
Fixed in dev with fdc9ac1b

closed 2023-10-04 by olorinmaia